gnuradio/gnuradio #3199: I need to display mic jack device name, but "arecord" doesn't exist in MacOS.

All3xJ, 2020-02-12:
I'm following this guide: [https://wiki.gnuradio.org/index.php/Audio\_Source](https://wiki.gnuradio.org/index.php/Audio_Source) and it says I need to get the device name (for example `hw:CARD=Device,DEV=0`). But arecord doesn't exist in macos. Any other way to achieve this?

michaelld, 2020-02-12:
Try executing the "dial_tone" example, generally installed into $PREFIX/share/gnuradio/examples/audio/audio_to_file.py" , with the arguments "-I foo foo" ... that will fail, but it will print out a list of available audio input devices on your Mac. For output devices, you can use "dial_tone.py".

The above stated, if you go into the System Preferences -> Audio -> Input, you can use the exact name of any of those devices; since most contain spaces, put the whole name in quotes, e.g., "MacBook Pro Microphone".

michaelld, 2020-02-12:
I went ahead and updated the audio page you noted, to include info for OSX. Thanks for pointing out the deficit! I'm closing this issue now. please reopen if need be and explain what's going on.

closed 2020-02-12 by michaelld