gnuradio/gnuradio #3209: Fix unicode handling in YAML files

marcusmueller, 2020-02-16:
5de869ec0b7ef660d052ce7c0b86cd9d5eaadc2d is a bad fix (we need to fix it, even English literature usually says "Mueller and Muller"), but it highlights an underlying issue.

japm48, 2020-04-10:
I just reverted 5de869e, and, apparently, it works on my machine (TM), no trace of #2985.

I suspect it's because of python 3 (more context: arch, Gnome 3.36, pyaml 20.4.0-1, python-gobject 3.36.0).

Proof:
[image]
[image]

marcusmueller, 2023-04-10:
seems to be solved

closed 2023-04-10 by marcusmueller