gnuradio/gnuradio #3215: Segmentation fault when adding qt instrumentation block

cafeclimber, 2020-02-17:
OS: Fedora 31
uname -a: Linux localhost.localdomain 5.4.18-200.fc31.x86_64 #1 SMP Fri Feb 7 14:50:05 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
Gnuradio build: maint-3.8 UHD 3.15

Running gnuradio-companion, I get a similar error to #2979 which was closed by the person who opened it but I don't see that they came to a resolution. I ran it through valgrind and this was the output I got. Let me know if there's anything else I can post to be helpful.

==6703== Memcheck, a memory error detector
==6703== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==6703== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==6703== Command: /home/bayleef/Documents/HRG/gnuradio/gr-maint/bin/gnuradio-companion
==6703== 
==6703== Conditional jump or move depends on uninitialised value(s)
==6703==    at 0x4B77707: PyUnicode_Decode (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B77990: PyUnicode_FromEncodedObject (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B8D856: ??? (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B75864: _PyObject_FastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B77278: ??? (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4BAD4CD: _PyEval_EvalFrameDefault (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B65939: _PyFunction_FastCallDict (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B3A8C5: _PyObject_Call_Prepend (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B749E2: ??? (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B758C6: _PyObject_FastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B77278: ??? (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4BAD4CD: _PyEval_EvalFrameDefault (in /usr/lib64/libpython3.7m.so.1.0)
==6703== 
==6703== Conditional jump or move depends on uninitialised value(s)
==6703==    at 0x136746CD: boost::filesystem::detail::directory_iterator_increment(boost::filesystem::directory_iterator&, boost::system::error_code*) (in /usr/lib64/libboost_filesystem.so.1.69.0)
==6703==    by 0x13674D0A: boost::filesystem::detail::directory_iterator_construct(boost::filesystem::directory_iterator&, boost::filesystem::path const&, boost::system::error_code*) (in /usr/lib64/libboost_filesystem.so.1.69.0)
==6703==    by 0x13180C39: directory_iterator (operations.hpp:905)
==6703==    by 0x13180C39: gr::prefs::_sys_prefs_filenames[abi:cxx11]() (prefs.cc:66)
==6703==    by 0x13180FF1: gr::prefs::prefs() (prefs.cc:53)
==6703==    by 0x131810F3: gr::prefs::singleton() (prefs.cc:48)
==6703==    by 0x12F425D0: _wrap_prefs_singleton (runtime_swigPYTHON_wrap.cxx:43550)
==6703==    by 0x4B44A89: _PyMethodDef_RawFastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B44D42: _PyCFunction_FastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B77202: ??? (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4BB1E2C: _PyEval_EvalFrameDefault (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B652D1: _PyFunction_FastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B770EE: ??? (in /usr/lib64/libpython3.7m.so.1.0)
==6703== 
==6703== Conditional jump or move depends on uninitialised value(s)
==6703==    at 0x136746CD: boost::filesystem::detail::directory_iterator_increment(boost::filesystem::directory_iterator&, boost::system::error_code*) (in /usr/lib64/libboost_filesystem.so.1.69.0)
==6703==    by 0x13180CCB: increment (operations.hpp:941)
==6703==    by 0x13180CCB: increment<boost::filesystem::directory_iterator> (iterator_facade.hpp:556)
==6703==    by 0x13180CCB: operator++ (iterator_facade.hpp:666)
==6703==    by 0x13180CCB: operator++<boost::filesystem::directory_iterator, boost::filesystem::directory_entry, boost::iterators::single_pass_traversal_tag, boost::filesystem::directory_entry&, long int> (iterator_facade.hpp:799)
==6703==    by 0x13180CCB: gr::prefs::_sys_prefs_filenames[abi:cxx11]() (prefs.cc:68)
==6703==    by 0x13180FF1: gr::prefs::prefs() (prefs.cc:53)
==6703==    by 0x131810F3: gr::prefs::singleton() (prefs.cc:48)
==6703==    by 0x12F425D0: _wrap_prefs_singleton (runtime_swigPYTHON_wrap.cxx:43550)
==6703==    by 0x4B44A89: _PyMethodDef_RawFastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B44D42: _PyCFunction_FastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B77202: ??? (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4BB1E2C: _PyEval_EvalFrameDefault (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B652D1: _PyFunction_FastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B770EE: ??? (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4BAD4CD: _PyEval_EvalFrameDefault (in /usr/lib64/libpython3.7m.so.1.0)
==6703== 
==6703== Conditional jump or move depends on uninitialised value(s)
==6703==    at 0x136746CD: boost::filesystem::detail::directory_iterator_increment(boost::filesystem::directory_iterator&, boost::system::error_code*) (in /usr/lib64/libboost_filesystem.so.1.69.0)
==6703==    by 0x13180CCB: increment (operations.hpp:941)
==6703==    by 0x13180CCB: increment<boost::filesystem::directory_iterator> (iterator_facade.hpp:556)
==6703==    by 0x13180CCB: operator++ (iterator_facade.hpp:666)
==6703==    by 0x13180CCB: operator++<boost::filesystem::directory_iterator, boost::filesystem::directory_entry, boost::iterators::single_pass_traversal_tag, boost::filesystem::directory_entry&, long int> (iterator_facade.hpp:799)
==6703==    by 0x13180CCB: gr::prefs::_sys_prefs_filenames[abi:cxx11]() (prefs.cc:68)
==6703==    by 0x13180FF1: gr::prefs::prefs() (prefs.cc:53)
==6703==    by 0x12F42633: _wrap_new_prefs (runtime_swigPYTHON_wrap.cxx:44471)
==6703==    by 0x4B44A89: _PyMethodDef_RawFastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B44D42: _PyCFunction_FastCallKeywords (in /usr/lib64/libpython3.7m.so.1.0)
==6703==    by 0x4B77202: ??? (in /usr/lib64/libpython3.7m.so.1.0)
[... 50 lines cut ...]
==6703==    by 0x40148FD: _dl_open (in /usr/lib64/ld-2.30.so)
==6703==    by 0x4DA739B: dlopen_doit (in /usr/lib64/libdl-2.30.so)
==6703==    by 0x49AA437: _dl_catch_exception (in /usr/lib64/libc-2.30.so)
==6703==    by 0x49AA502: _dl_catch_error (in /usr/lib64/libc-2.30.so)
==6703==    by 0x4DA7B08: _dlerror_run (in /usr/lib64/libdl-2.30.so)
==6703==    by 0x4DA7429: dlopen@@GLIBC_2.2.5 (in /usr/lib64/libdl-2.30.so)
==6703== 
==6703== HEAP SUMMARY:
==6703==     in use at exit: 92,291,074 bytes in 720,848 blocks
==6703==   total heap usage: 2,696,456 allocs, 1,975,608 frees, 438,056,167 bytes allocated
==6703== 
==6703== LEAK SUMMARY:
==6703==    definitely lost: 49,650 bytes in 208 blocks
==6703==    indirectly lost: 636,589 bytes in 6,146 blocks
==6703==      possibly lost: 25,118,135 bytes in 241,649 blocks
==6703==    still reachable: 65,598,556 bytes in 466,388 blocks
==6703==                       of which reachable via heuristic:
==6703==                         length64           : 14,192 bytes in 197 blocks
==6703==                         newarray           : 3,072 bytes in 112 blocks
==6703==         suppressed: 0 bytes in 0 blocks
==6703== Rerun with --leak-check=full to see details of leaked memory
==6703== 
==6703== Use --track-origins=yes to see where uninitialised values come from
==6703== For lists of detected and suppressed errors, rerun with: -s
==6703== ERROR SUMMARY: 39 errors from 4 contexts (suppressed: 0 from 0)

velichkov, 2020-02-19:
Hi @cafeclimber,

> Gnuradio build: maint-3.8 UHD 3.15

How did you install gnuradio - you built it from source, using pybombs or from the [copr](https://copr.fedorainfracloud.org/coprs/gnuradio/gnuradio-releases/) repository?

> adding qt instrumentation block

Is this a custom OOT block?

If yes make sure that your `.so` files does not have `undefined symbols`
```
cd build
ldd -r ./lib/*.so
ldd -r ./swig/*swig.so
```
And if this is not a custom block can you provide a flowgraph?

cafeclimber, 2020-02-24:
Hi, I built it using pybombs. I've not installed any custom blocks, only uhd and gnuradio

velichkov, 2020-02-24:
> I built it using pybombs. 

I can't reproduce this crash but I'm using the [copr](https://copr.fedorainfracloud.org/coprs/gnuradio/gnuradio-releases/) version. In case you don't need the Qt4 you may try removing it.

> I've not installed any custom blocks, only uhd and gnuradio

Do you have any gnuradio pakcages installed from the Fedora repostiory? If yes then try removing them as having multiple versions installed at the same time could cause various problems.

> ==6703== at 0x3FB74E88: ??? (in /usr/lib64/libQtGui.so.4.8.7)
> ==6703== by 0x4010F29: call_init.part.0 (in /usr/lib64/ld-2.30.so)
> ==6703== by 0x4011030: _dl_init (in /usr/lib64/ld-2.30.so)

Can you provide a list of all Qt4 packages installed on your system - `rpm -qa | grep qt | grep "4\.8"` ?

Also when does it crash - when you add the QT block or when you execute the flowgraph?

cafeclimber, 2020-02-24:
I haven't installed any gnuradio packages from the fedora repositories. This was a fresh install of Fedora on which I immediately installed gnuradio with PyBOMBS. 

It crashes when I add the block.

Here is the output from that command:
```
$ rpm -qa | grep qt | grep "4\.8"                                                 
qt-x11-4.8.7-50.fc31.x86_64
qt-common-4.8.7-50.fc31.noarch
qt-devel-4.8.7-50.fc31.x86_64
qt-4.8.7-50.fc31.x86_64
```

velichkov, 2020-02-24:
> I haven't installed any gnuradio packages from the fedora repositories. This was a fresh install of Fedora on which I immediately installed gnuradio with PyBOMBS.

OK. Where did you install gnuradio - `/usr/local` or a custom prefix? Can you provide the output of
```
find /usr -name libgnuradio-qtgui.so 2>/dev/null -exec ldd -r {} \;
```
in case you've installed in a custom prefix replace `/usr/` with the full path to your `prefix`.

> It crashes when I add the block.

OK. Does it crash with all QT blocks or some of them? Give me the exact names of the blocks you have tried with and it crashed.

> Here is the output from that command:
> 
> ```
> $ rpm -qa | grep qt | grep "4\.8"                                                 
> qt-x11-4.8.7-50.fc31.x86_64
> qt-common-4.8.7-50.fc31.noarch
> qt-devel-4.8.7-50.fc31.x86_64
> qt-4.8.7-50.fc31.x86_64
> ```

Qt4 has been removed from a lot of Linux distros or it's currently being removed. Do you need it and are you willing to try removing all these packages?

In case you need it for some other application, can you try removing the `qt-devel` package and then rebuild and reinstall gnuradio?

cafeclimber, 2020-02-25:
Sure thing.

```
$ find /usr -name libgnuradio-qtgui.so 2>/dev/null -exec ldd -r {} \;
	linux-vdso.so.1 (0x00007fff821c9000)
	libgnuradio-filter.so.3.8.0 => /home/bayleef/Documents/HRG/gnuradio/gr-maint/src/gnuradio/build/gr-filter/lib/libgnuradio-filter.so.3.8.0 (0x00007f13fbfc0000)
	libQt5Widgets.so.5 => /lib64/libQt5Widgets.so.5 (0x00007f13fb8f0000)
	libgnuradio-fft.so.3.8.0 => /home/bayleef/Documents/HRG/gnuradio/gr-maint/src/gnuradio/build/gr-fft/lib/libgnuradio-fft.so.3.8.0 (0x00007f13fb8b8000)
	libfftw3f.so.3 => /lib64/libfftw3f.so.3 (0x00007f13fb6a3000)
	libfftw3f_threads.so.3 => /lib64/libfftw3f_threads.so.3 (0x00007f13fb699000)
	libgnuradio-blocks.so.3.8.0 => /home/bayleef/Documents/HRG/gnuradio/gr-maint/src/gnuradio/build/gr-blocks/lib/libgnuradio-blocks.so.3.8.0 (0x00007f13fb407000)
	libgnuradio-runtime.so.3.8.0 => /home/bayleef/Documents/HRG/gnuradio/gr-maint/src/gnuradio/build/gnuradio-runtime/lib/libgnuradio-runtime.so.3.8.0 (0x00007f13fb30b000)
	libgnuradio-pmt.so.3.8.0 => /home/bayleef/Documents/HRG/gnuradio/gr-maint/src/gnuradio/build/gnuradio-runtime/lib/pmt/libgnuradio-pmt.so.3.8.0 (0x00007f13fb2a4000)
	libboost_program_options.so.1.69.0 => /lib64/libboost_program_options.so.1.69.0 (0x00007f13fb21c000)
	libboost_filesystem.so.1.69.0 => /lib64/libboost_filesystem.so.1.69.0 (0x00007f13fb1ff000)
	libboost_system.so.1.69.0 => /lib64/libboost_system.so.1.69.0 (0x00007f13fb1fa000)
	libboost_regex.so.1.69.0 => /lib64/libboost_regex.so.1.69.0 (0x00007f13fb0ff000)
	libboost_thread.so.1.69.0 => /lib64/libboost_thread.so.1.69.0 (0x00007f13fb0d1000)
	libboost_chrono.so.1.69.0 => /lib64/libboost_chrono.so.1.69.0 (0x00007f13fb0c2000)
	libboost_date_time.so.1.69.0 => /lib64/libboost_date_time.so.1.69.0 (0x00007f13fb0ae000)
	libboost_atomic.so.1.69.0 => /lib64/libboost_atomic.so.1.69.0 (0x00007f13fb0a7000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f13fb085000)
	liblog4cpp.so.5 => /lib64/liblog4cpp.so.5 (0x00007f13fb040000)
	libgmpxx.so.4 => /lib64/libgmpxx.so.4 (0x00007f13fb037000)
	libgmp.so.10 => /home/bayleef/Documents/HRG/gnuradio/gr-maint/lib/libgmp.so.10 (0x00007f13fafc3000)
	librt.so.1 => /lib64/librt.so.1 (0x00007f13fafb8000)
	libvolk.so.2.0 => /home/bayleef/Documents/HRG/gnuradio/gr-maint/src/gnuradio/build/volk/lib/libvolk.so.2.0 (0x00007f13facc1000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f13facba000)
	liborc-0.4.so.0 => /lib64/liborc-0.4.so.0 (0x00007f13fac38000)
	libqwt.so.6 => /lib64/libqwt.so.6 (0x00007f13faae1000)
	libQt5Gui.so.5 => /lib64/libQt5Gui.so.5 (0x00007f13fa5a2000)
	libQt5Core.so.5 => /lib64/libQt5Core.so.5 (0x00007f13fa06e000)
	libpython3.7m.so.1.0 => /lib64/libpython3.7m.so.1.0 (0x00007f13f9d20000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f13f9b26000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f13f99e0000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f13f99c6000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f13f97fd000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f13fc1f2000)
	libicudata.so.63 => /lib64/libicudata.so.63 (0x00007f13f7e0a000)
	libicui18n.so.63 => /lib64/libicui18n.so.63 (0x00007f13f7b27000)
	libicuuc.so.63 => /lib64/libicuuc.so.63 (0x00007f13f7954000)
	libQtSvg.so.4 => /lib64/libQtSvg.so.4 (0x00007f13f78f3000)
	libQtOpenGL.so.4 => /lib64/libQtOpenGL.so.4 (0x00007f13f77db000)
	libQtGui.so.4 => /lib64/libQtGui.so.4 (0x00007f13f6bf6000)
	libQtCore.so.4 => /lib64/libQtCore.so.4 (0x00007f13f68c5000)
	libGL.so.1 => /lib64/libGL.so.1 (0x00007f13f6831000)
	libpng16.so.16 => /lib64/libpng16.so.16 (0x00007f13f67fb000)
	libz.so.1 => /lib64/libz.so.1 (0x00007f13f67e1000)
	libharfbuzz.so.0 => /lib64/libharfbuzz.so.0 (0x00007f13f66e7000)
	libsystemd.so.0 => /lib64/libsystemd.so.0 (0x00007f13f6634000)
	libpcre2-16.so.0 => /lib64/libpcre2-16.so.0 (0x00007f13f65af000)
	libglib-2.0.so.0 => /lib64/libglib-2.0.so.0 (0x00007f13f6487000)
	libutil.so.1 => /lib64/libutil.so.1 (0x00007f13f6482000)
	libfreetype.so.6 => /lib64/libfreetype.so.6 (0x00007f13f63bf000)
	libXrender.so.1 => /lib64/libXrender.so.1 (0x00007f13f63b2000)
	libX11.so.6 => /lib64/libX11.so.6 (0x00007f13f626c000)
	libgobject-2.0.so.0 => /lib64/libgobject-2.0.so.0 (0x00007f13f6210000)
	libSM.so.6 => /lib64/libSM.so.6 (0x00007f13f6205000)
	libICE.so.6 => /lib64/libICE.so.6 (0x00007f13f61e7000)
	libXi.so.6 => /lib64/libXi.so.6 (0x00007f13f61d3000)
	libXrandr.so.2 => /lib64/libXrandr.so.2 (0x00007f13f61c6000)
	libXfixes.so.3 => /lib64/libXfixes.so.3 (0x00007f13f61bd000)
	libXcursor.so.1 => /lib64/libXcursor.so.1 (0x00007f13f61b0000)
	libXinerama.so.1 => /lib64/libXinerama.so.1 (0x00007f13f61ab000)
	libfontconfig.so.1 => /lib64/libfontconfig.so.1 (0x00007f13f6163000)
	libXext.so.6 => /lib64/libXext.so.6 (0x00007f13f614c000)
	libGLX.so.0 => /lib64/libGLX.so.0 (0x00007f13f6118000)
	libGLdispatch.so.0 => /lib64/libGLdispatch.so.0 (0x00007f13f605b000)
	libgraphite2.so.3 => /lib64/libgraphite2.so.3 (0x00007f13f6036000)
	liblzma.so.5 => /lib64/liblzma.so.5 (0x00007f13f600d000)
[... 82 lines cut ...]
	libuuid.so.1 => /lib64/libuuid.so.1 (0x00007f39b9e13000)
	libexpat.so.1 => /lib64/libexpat.so.1 (0x00007f39b9de5000)
	libgpg-error.so.0 => /lib64/libgpg-error.so.0 (0x00007f39b9dc0000)
	libXau.so.6 => /lib64/libXau.so.6 (0x00007f39b9dba000)
```

> OK. Does it crash with all QT blocks or some of them? Give me the exact names of the blocks you have tried with and it crashed.

It crashes with every block in the Instrumentation -> QT Dropdown:
    QT GUI Bercurve Sink
    QT GUI Constellation Sink
    QT GUI Frequency Sink
[... cut ...] togram Sink
    QT GUI Number Sink
    QT GUI Sink
    QT GUI Time Raster Sink
    QT GUI Time Sink
    QT GUI Vector Sink
    QT GUI Waterfall Sink

It seems to be only this blocks. I've tried with other blocks, including those in the GUI Widgets -> QT dropdown with no issues.

> In case you need it for some other application, can you try removing the qt-devel package and then rebuild and reinstall gnuradio?

Interesting, I just removed the qt-devel package (no others) and the issue is resolved

closed 2020-02-25 by cafeclimber

velichkov, 2020-02-25:
> ```
> $ find /usr -name libgnuradio-qtgui.so 2>/dev/null -exec ldd -r {} \;
> 	linux-vdso.so.1 (0x00007fff821c9000)
> 	libgnuradio-filter.so.3.8.0 => /home/bayleef/Documents/HRG/gnuradio/gr-maint/src/gnuradio/build/gr-filter/lib/libgnuradio-filter.so.3.8.0 (0x00007f13fbfc0000)
>```

So you have installed gnuradio in the `/home/bayleef/Documents/HRG/gnuradio/gr-maint/` prefix but you _did not_ changed the `/usr/` with this prefix in the above command. Are you absolutely sure that there are not gnuradio packages installed in `/usr/`?

Can you provide the output of ` find /usr -name libgnuradio-qtgui.so`?

>```
> 	libQt5Widgets.so.5 => /lib64/libQt5Widgets.so.5 (0x00007f13fb8f0000)
> 	libQt5Gui.so.5 => /lib64/libQt5Gui.so.5 (0x00007f13fa5a2000)
> 	libQt5Core.so.5 => /lib64/libQt5Core.so.5 (0x00007f13fa06e000)
> 	libQtSvg.so.4 => /lib64/libQtSvg.so.4 (0x00007f13f78f3000)
> 	libQtOpenGL.so.4 => /lib64/libQtOpenGL.so.4 (0x00007f13f77db000)
> 	libQtGui.so.4 => /lib64/libQtGui.so.4 (0x00007f13f6bf6000)
> 	libQtCore.so.4 => /lib64/libQtCore.so.4 (0x00007f13f68c5000)
> 	libQt5Widgets.so.5 => /lib64/libQt5Widgets.so.5 (0x00007f39bf8ff000)
> 	libQt5Gui.so.5 => /lib64/libQt5Gui.so.5 (0x00007f39be5b1000)
> 	libQt5Core.so.5 => /lib64/libQt5Core.so.5 (0x00007f39be07d000)
> 	libQtSvg.so.4 => /lib64/libQtSvg.so.4 (0x00007f39bb902000)
> 	libQtOpenGL.so.4 => /lib64/libQtOpenGL.so.4 (0x00007f39bb7ec000)
> 	libQtGui.so.4 => /lib64/libQtGui.so.4 (0x00007f39bac07000)
> 	libQtCore.so.4 => /lib64/libQtCore.so.4 (0x00007f39ba8d4000)
>```

The Qt4 libraries gets loaded together with Qt5 libraries and that's the most probable reason for this segfault.

> It crashes with every block in the Instrumentation -> QT Dropdown:
> QT GUI Bercurve Sink
> QT GUI Constellation Sink
> QT GUI Frequency Sink
> QT GUI Histogram Sink
> QT GUI Number Sink
> QT GUI Sink
> QT GUI [...]

That's good to know.

> > In case you need it for some other application, can you try removing the qt-devel package and then rebuild and reinstall gnuradio?
> 
> Interesting, I just removed the qt-devel package (no others) and the issue is resolved

Did you rebuild and reinstall `gnuradio` after removing the `qt-devel` package or just removing it without rebuilding fixed the problem?

cafeclimber, 2020-02-27:
> So you have installed gnuradio in the /home/bayleef/Documents/HRG/gnuradio/gr-maint/ prefix but you did not changed the /usr/ with this prefix in the above command. Are you absolutely sure that there are not gnuradio packages installed in /usr/?

Sorry, I just copied the command you gave when I posted it here, but when I ran it, I did substitute /usr with my prefix directory.

> Did you rebuild and reinstall gnuradio after removing the qt-devel package or just removing it without rebuilding fixed the problem?

I did not rebuild and reinstall. I only removed the qt-devel package with my package manager which fixed the problem.

velichkov, 2020-02-27:
> Sorry, I just copied the command you gave when I posted it here, but when I ran it, I did substitute /usr with my prefix directory.

OK. 

> > Did you rebuild and reinstall gnuradio after removing the qt-devel package or just removing it without rebuilding fixed the problem?
> 
> I did not rebuild and reinstall. I only removed the qt-devel package with my package manager which fixed the problem.

OK. It's really useful to know that no rebuilding is needed. Most probably a wrong `.so` is being dynamically loaded with `dlopen` and this result in the segfault.