gnuradio/gnuradio #3216: problem in flowgraph creation in gnuradio 3.8 on windows

FONQRI, 2020-02-17:
Hi,
I tried to use uhd_fft.grc example of gnuradio 3.8 in example folder and i got some errors and i figured out how to fix them but their is problem in changing gain that when i change it in ui it dose not effect 
here is my grc and top_block i tried same grc in linux and it work correctly and difference is in function set_gain that in linux its 3 lines and last line is 
```c++
self.uhd_usrp_source_0.set_gain(self.gain)
```
[uhd_gnu_example.zip](https://github.com/gnuradio/gnuradio/files/4215608/uhd_gnu_example.zip)
 i think there is a problem in flowgraph generation in windows.
Note : I use gnuradio v3.8.0.0/v1.7 BETA in windows.
I think that's a bug.

michaelld, 2020-02-17:
Can you try the recently released 3.8.1.0-RC1 ? it has a -ton- of fixes related to 3.8.0.0.

michaelld, 2020-02-17:
For example and relevant to your query: https://github.com/gnuradio/gnuradio/issues/2791

FONQRI, 2020-02-17:
> Can you try the recently released 3.8.1.0-RC1 ? it has a -ton- of fixes related to 3.8.0.0.

Is there any compiled release for windows ? like v3.8.0.0/v1.7 BETA ?

michaelld, 2020-02-17:
I can't find one. The 3.8.1.0-RC1 is really recent ... likely as not it's going to be a week or more before such a binary comes up. I'd encourage you to query the "discuss-gnuradio" list & maybe the Windows-binary folks will create one.

FONQRI, 2020-02-18:
Ok i'll try to build windows version to see if it's solved.

nickoe, 2020-03-30:
@FONQRI  I tried to download your example and generate the py file. The only difference I saw was that yours were generated with python2 and prepended a u on a few strings as expected. 

No difference around the set_gain line you mentioned. Maybe show us those three lines you mention.

FONQRI, 2020-04-06:
> @FONQRI I tried to download your example and generate the py file. The only difference I saw was that yours were generated with python2 and prepended a u on a few strings as expected.
> 
> No difference around the set_gain line you mentioned. Maybe show us those three lines you mention.

Here is windows and linux top_blocks 
[GNU_FLOW.zip](https://github.com/gnuradio/gnuradio/files/4438622/GNU_FLOW.zip)

nickoe, 2020-04-06:
Ok, I see. A quick search of `uhd_usrp_source_` in this repository via github leads me to a similar issue https://github.com/gnuradio/gnuradio/issues/2791.

This was fixed via https://github.com/gnuradio/gnuradio/pull/2821 on master and included in this PR for maint-3.8 https://github.com/gnuradio/gnuradio/pull/2994 on the 4th of Jan. This means that it is included in the v3.8.1.0-rc1 tag.

Do we know the exact version (commit sha1) used for the windows build?

closed 2020-04-06 by nickoe