gnuradio/gnuradio #3217: ZMQ Message Source blocks fail to receive data from non-GRC sources

duggabe, 2020-02-19:
In trying to use the ZMQ PULL Message Source block to communicate with a separate, non-GRC Python process, the PULL Message Source block fails to receive any data. A series of tests have been performed to validate the condition. Similar tests were performed using REQ/REP and PUB/SUB sockets. They also failed. Note that only interactions between GRC and non-GRC sources fail.

platform: Raspberry Pi 3B+
OS: 2020-02-13-raspbian-buster-full
GR: 3.8.1.1-rc1
GRC: 3.8.1.0
Current libzmq version is 4.3.1
Current  pyzmq version is 18.1.1

The following files are used in the tests below:
msg_test.py -    https://pastebin.com/wabF6Tsz
msg_test2.py -  https://pastebin.com/e3TE7dcb
zmqPULL.py -   https://pastebin.com/p93GGbtu
zmqPUSH.py -  https://pastebin.com/hdUkhysV

Test case 1: GRC from GRC
- terminal A:
-   execute 'python3 -u msg_test.py'
- terminal B:
-   execute 'python3 -u msg_test2.py'
- terminal A displays "TEST" once per second.
- stop both programs by clicking the X in the GUI windows

Test case 2: Python from Python
- terminal A:
-   execute 'python3 zmqPULL.py'
- terminal B:
-   execute 'python3 zmqPUSH.py'
- terminal A displays the date and time once per second.
- stop both programs with Control-C

Test case 3: GRC from Python
- terminal A:
-   execute 'python3 -u msg_test.py'
- terminal B:
-   execute 'python3 zmqPUSH.py'
- terminal A *should* display the date and time once per second, but it FAILS: there is no output.
- stop terminal A by clicking the X in the GUI window
- stop terminal B with Control-C

talih0, 2020-02-19:
@duggabe 
The pull source block connects by default (https://github.com/gnuradio/gnuradio/blob/master/gr-zeromq/lib/pull_msg_source_impl.cc#L48)

Your python script is also connecting on the push side. 
If you change your python script to bind on the push side, it will work.

talih0, 2020-02-19:
@duggabe 
Also, If the zmq pull source is setup to read gr_complex, it is expecting to receive sizeof(gr_complex)*num_symbols bytes. 
This is likely not to be satisfied if you send a general string, and gnuradio will throw an exception.

velichkov, 2020-02-19:
Hi @duggabe,

In addtion to the client/server problem the `ZMQ Message Source` block expects a valid `pmt` messages

https://github.com/gnuradio/gnuradio/blob/45af34ed129b46a28c1f6d8ec0ea27a29411460d/gr-zeromq/lib/pull_msg_source_impl.cc#L98-L107

while `zmqPUSH.py` is sending plain strings 

```python
    _tim = datetime.datetime.now()
    _out = str(_tim)
    message = str.encode (_out)
    try:
        sock.send (message)
```

and `pmt::deserialize` throws an exception.

```
$ python3 ./msg_test.py
terminate called after throwing an instance of 'pmt::exception'
  what():  pmt::deserialize: malformed input stream, tag value = : 50
Aborted (core dumped)
```

The `pmt` encoding is a custom binary encoding so for now you probably need to import the guradio's `pmt` module in your scripts and use the `pmt.serialize_str`, `pmt.deserialize_str` and `pmt.to_pmt` functions.

```python
import pmt
sock.send (pmt.serialize_str(pmt.to_pmt('Hello World!')))
```

devnulling, 2020-02-20:
@duggabe  combining what @talih0 and @velichkov have recommend this works:

```
# zmqPUSH.py

import zmq
import datetime
import time
import signal
import pmt

print("Current libzmq version is %s" % zmq.zmq_version())
print("Current  pyzmq version is %s" % zmq.__version__)

context = zmq.Context()

# create socket
sock = context.socket(zmq.PUSH)
sock.bind("tcp://127.0.0.1:5558")

while True:
    _tim = datetime.datetime.now()
    _out = str(_tim)
    message = str.encode (_out)
    try:
        #sock.send (message)
        sock.send (pmt.serialize_str(pmt.to_pmt('Hello World!')))
        print('sending....')
        time.sleep(1)
    except KeyboardInterrupt:
        print ("interrupt received. shutting down.")
        # clean up
        sock.close()
        context.term()
        exit()
```

[zmq_msg_in.grc.txt](https://github.com/gnuradio/gnuradio/files/4228422/zmq_msg_in.grc.txt)

duggabe, 2020-02-20:
@devnulling Yes, thank you and the others for your help. It works for me also. And using my time message works as well:
<pre>
while True:
    _tim = datetime.datetime.now()
    _out = str (_tim)
    try:
        sock.send (pmt.serialize_str(pmt.to_pmt(_out)))
        # print('sending....')
        time.sleep(1)
</pre>

Now that I know what works, I will produce documentation which will help others!

closed 2020-02-20 by duggabe