gnuradio/gnuradio #3219: Ubuntu binary install for 3.8 on ARM is (accidentally?) delivering 3.7

noahhsmith2, 2020-02-20:
Hi all,

Below is my first thoughts yesterday (on the slack channel) - in a nutshell - suspect that the ARM build in ppa:gnuradio/gnuradio-releases is accidentally using the 3.7 branch rather than the 3.8 branch...

An install on raspi from ppa:gnuradio/gnuradio-releases gets 3.7 (an install on x86 from ppa:gnuradio/gnuradio-releases gets 3.8, there'd be plenty of holler'n if not!:)

hi all - quick question about ARM packages and ubuntu
seems there are ARM packages for 3.9 (ppa:gnuradio/gnuradio-master)
and for 3.7 (ppa:gnuradio/gnuradio-releases-3.7)
but not for 3.8 (ppa:gnuradio/gnuradio-releases) ?
i've been able to install 3.9 and 3.7 to raspi - but seems i need 3.8 for my blocks (having troubles with 3.9)...
are there 3.8 ARM packages out there somewhere? :smiley:
ahha should probly add that if you do an ARM install from ppa:gnuradio/gnuradio-releases you do get something - 3.7! :)
thinkn about it - can hazard a guess - could the ARM build for ppa:gnuradio/gnuradio-releases have accidentally used the 3.7 branch instead of the 3.8 branch...?... 

Cheers,
Noah

dkozel, 2020-03-01:
Hi @noahhsmith2 

Thanks for the report. I've just taken a quick look at the latest `arm64` package and it's (almost certainly) shipping 3.8.1.0. If there's something wrong with the package that's subtler than what I checked then no blocks would show up.

What exactly have you observed? Do you still have the system and problem?

noahhsmith2, 2020-03-01:
Hi Derek, awesome, thanks, yep still have the raspi sitting on my office
desk and will try a ppa install tomorrow and see what happens!:)

What I can say is that two weeks ago, when I was getting 3.8 (.0 I'm pretty
sure, see below) on x86, I got 3.7.x on the raspi, with the same basic
ubuntu state... Man I hope it works these binary installs to ubuntu from
ppa ROCK!!! Have you ever built gnuradio on a raspi?:) Laugh, it works,
it's old school, just gotta be patient:)

I suspect the switch to 3.8.1 came this last week cause my flowgraphs all
magically broke and I had to rebuild this weekend;)

More tomorrow, cheers,
Noah

noahhsmith2, 2020-03-02:
Hi Derek, it worked great this morning - made a fresh ubuntu sd card and install of 3.8 on raspi worked perfectly, exactly as in the instructions!

Super - thanks so much for checking in on this. No idea what was going wrong a few weeks ago, but I burned plenty of time trying to figure it out, and ended up building from source...

Now we are all set - fully automated ppa install of 3.8.1 on raspi with fresh ubuntu

ubuntu@ubuntu:~$ history
    1  sudo apt purge gnuradio
    2  sudo add-apt-repository ppa:gnuradio/gnuradio-releases
    3  sudo apt install gnuradio
    4  python3
ubuntu@ubuntu:~$ python3
Python 3.7.5 (default, Nov 20 2019, 09:21:52) 
[GCC 9.2.1 20191008] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gnuradio import gr
>>> gr.version()
'3.8.1.0'

Cheers,
Noah

closed 2020-03-02 by noahhsmith2