gnuradio/gnuradio #3220: Missing deconstructor in ZMQ_PUSH_Sink

Dur0k, 2020-02-20:
Due to a missing deconstructror in gnuradio/gr-zeromq/lib/push_sink_impl.cc the used port is not released after "deleting" block object with pythons del function.

After adding 
```
push_sink_impl::~push_sink_impl()
{
}
```
to  gnuradio/gr-zeromq/lib/push_sink_impl.cc and
```
~push_sink_impl();
```
to gnuradio/gr-zeromq/lib/push_sink_impl.h the port is correctly released and can be bound again.

This occurred while trying to disconnect, delete and afterwards create a new ZMQ_PUSH_Sink on the same port. The flowgraph was started externally in python via class object of the flowgraph.

talih0, 2020-02-20:
The compiler adds a default destructor automatically if the user does not create one.

You should probably instead test setting linger time to 0 as is done in the push_msg_sink code:

https://github.com/gnuradio/gnuradio/blob/master/gr-zeromq/lib/push_msg_sink_impl.cc#L44

michaelld, 2020-02-21:
> The compiler adds a default destructor automatically if the user does not create one.

This statement isn't _necessarily_ true or correct. In some recent C++ standards, if a destructor isn't supplied the compiler might or not create a virtual default. This case will generally result in a compiler warning during build, but might result in a build failure. If the code does compile, then it stands a decent chance of runtime failure. In my testing, such issues happen more with Clang and GCC; no idea why.

In this specific case, adding in the virtual destructor is the wise thing to do, if nothing else to avoid such situations in the future (e.g., moving to C++14 or C++17 or C++20).

michaelld, 2020-02-21:
@Dur0k do you feel comfortable creating a PR for this issue? If you do, please check _all_ of the ZMQ sink/source blocks in gr-zeromq & fix any that need this tweak.

marcusmueller, 2024-01-08:
there's a virtual destructor in `gr::zeromq::base_impl`, so that must get called. Which at least shuts down the context, closes the socket.

However, usually dtor bubbling happens at the end of a destructor call. 

So, will need to reproduce to make sure I'm understanding which dtors really get called here.