gnuradio/gnuradio #3227: gnuradio libqt4 crash for opensuse Tumbleweed

kk-1, 2020-02-21:
Hello,
I am using openSUSE Tumbleweed and gnuradio 3.8
It crashes when I try to open example files.
I do not know where is the problem. Please let me know if you need more info.
With the hope that they will be useful for you, here are dmesg lines:
...
373.579889] show_signal_msg: 59 callbacks suppressed
[  373.579891] python3[9353]: segfault at 7fb7b983a390 ip 00007fb7ae2073f4 sp 00007ffccf2283d0 error 7 in libQtGui.so.4.8.7[7fb7ae18a000+6e8000]
[  373.579896] Code: 07 0f 95 c0 48 8b 05 1b 6b 91 00 c6 05 cc ef 91 00 00 c7 05 c6 ef 91 00 00 00 00 00 c6 05 c3 ef 91 00 00 48 89 05 c4 ef 91 00 <f0> ff 00 0f 95 c2 f0 ff 0f 0f 95 c0 84 c0 74 1e 48 8d 15 d5 9a 91
[  373.634104] python3[9382]: segfault at 7f6b57353390 ip 00007f6b544a53f4 sp 00007ffd27c660c0 error 7 in libQtGui.so.4.8.7[7f6b54428000+6e8000]
[  373.634108] Code: 07 0f 95 c0 48 8b 05 1b 6b 91 00 c6 05 cc ef 91 00 00 c7 05 c6 ef 91 00 00 00 00 00 c6 05 c3 ef 91 00 00 48 89 05 c4 ef 91 00 <f0> ff 00 0f 95 c2 f0 ff 0f 0f 95 c0 84 c0 74 1e 48 8d 15 d5 9a 91
[  373.684989] python3[9384]: segfault at 7f3d7ce8d390 ip 00007f3d79fdf3f4 sp 00007ffd18cd30c0 error 7 in libQtGui.so.4.8.7[7f3d79f62000+6e8000]
[  373.684992] Code: 07 0f 95 c0 48 8b 05 1b 6b 91 00 c6 05 cc ef 91 00 00 c7 05 c6 ef 91 00 00 00 00 00 c6 05 c3 ef 91 00 00 48 89 05 c4 ef 91 00 <f0> ff 00 0f 95 c2 f0 ff 0f 0f 95 c0 84 c0 74 1e 48 8d 15 d5 9a 91
[  373.735288] python3[9386]: segfault at 7f9e0cd63390 ip 00007f9e09eb53f4 sp 00007ffcd765b120 error 7 in libQtGui.so.4.8.7[7f9e09e38000+6e8000]
[  373.735292] Code: 07 0f 95 c0 48 8b 05 1b 6b 91 00 c6 05 cc ef 91 00 00 c7 05 c6 ef 91 00 00 00 00 00 c6 05 c3 ef 91 00 00 48 89 05 c4 ef 91 00 <f0> ff 00 0f 95 c2 f0 ff 0f 0f 95 c0 84 c0 74 1e 48 8d 15 d5 9a 91
[  373.789077] python3[9388]: segfault at 7f5672a65390 ip 00007f566fbb73f4 sp 00007fffc15274e0 error 7 in libQtGui.so.4.8.7[7f566fb3a000+6e8000]
[  373.789081] Code: 07 0f 95 c0 48 8b 05 1b 6b 91 00 c6 05 cc ef 91 00 00 c7 05 c6 ef 91 00 00 00 00 00 c6 05 c3 ef 91 00 00 48 89 05 c4 ef 91 00 <f0> ff 00 0f 95 c2 f0 ff 0f 0f 95 c0 84 c0 74 1e 48 8d 15 d5 9a 91
[  396.437619] gnuradio-compan[9325]: segfault at 7fc0d5e71390 ip 00007fc071edb3f4 sp 00007ffd22033760 error 7 in libQtGui.so.4.8.7[7fc071e5e000+6e8000]
[  396.437622] Code: 07 0f 95 c0 48 8b 05 1b 6b 91 00 c6 05 cc ef 91 00 00 c7 05 c6 ef 91 00 00 00 00 00 c6 05 c3 ef 91 00 00 48 89 05 c4 ef 91 00 <f0> ff 00 0f 95 c2 f0 ff 0f 0f 95 c0 84 c0 74 1e 48 8d 15 d5 9a 91
...

marcusmueller, 2020-02-21:
Hi! Thanks for the report?

* How did you install GNU Radio?
* What example files specifically? (one specimen suffices)
* What specifically does "Open example files" entail, specifically?
  * Opening the GRC flow graph in the GNU Radio companion, OR
  * running a python GNU Radio program?

If you built GNU Radio yourself:

* Why did you use Qt4 on a modern distro?
* Does the same issue arise with Qt5 instead?

kk-1, 2020-02-21:
Thanks for the reply

I removed all the libqt4 rpms I had and now I can open examples without any crash!!!
Even run them!

On my system, I had libQT 5.14 and also libqt 4.8.7 both.
This was the problem I believe. I forgot why I installed libqt4 packages.
But the package manager did not complain when I removed them.

Below is useless but anyway, coming to your questions:

How did you install GNU Radio?
I installed the rpm package for ver3.8 from the repo:
http://download.opensuse.org/repositories/hamradio/openSUSE_Tumbleweed/
I remember trying other repos too

What is the output of gnuradio-config-info --version
>gnuradio-config-info --version
3.8.0.0

What example files specifically? (one specimen suffices)
Two different examples:
Loading: "/usr/share/gnuradio/examples/rds/rds_rx.grc"
Segmentation fault (core dumped)

Loading: "/usr/lib64/gnuradio/examples/audio/dial_tone.grc"
Segmentation fault (core dumped)

What specifically does "Open example files" entail, specifically?
Basically, on my system, I could not open any .grc file
from the GUI of "gnuradio-companion" executable.
GUI comes up but when I select "File->Open" core dumps!

Hope these helps.

closed 2020-02-21 by kk-1