gnuradio/gnuradio #3233: Gtk 3.0 version for gnuradio 3.8.0.0?

ChristopherRabotin, 2020-02-24:
I'm running Opensuse Tumbleweed with gnu-radio version 3.8.0.0 and it seems like I've got Gtk version 3.24.13 as well.

```
$ gnuradio-config-info -v
3.8.0.0
```

I get a Python API error when running GR companion:
```
 gnuradio-companion
Traceback (most recent call last):
  File "/usr/lib64/python3.7/site-packages/gnuradio/grc/gui/Application.py", line 102, in do_activate
    self.main_window = MainWindow(self, self.platform)
  File "/usr/lib64/python3.7/site-packages/gnuradio/grc/gui/MainWindow.py", line 108, in __init__
    self.console = Console()
  File "/usr/lib64/python3.7/site-packages/gnuradio/grc/gui/Console.py", line 44, in __init__
    self.text_display = TextDisplay()
  File "/usr/lib64/python3.7/site-packages/gnuradio/grc/gui/Dialogs.py", line 62, in __init__
    SimpleTextDisplay.__init__(self, text)
  File "/usr/lib64/python3.7/site-packages/gnuradio/grc/gui/Dialogs.py", line 44, in __init__
    self.set_text(text)
TypeError: Gtk.TextBuffer.set_text() takes exactly 3 arguments (2 given)
```

What version of Gtk should I be using? Thanks

velichkov, 2020-02-24:
Hi @ChristopherRabotin,

>  File "/usr/lib64/python3.7/site-packages/gnuradio/grc/gui/Dialogs.py", line 44, in __init__
>    self.set_text(text)
> TypeError: Gtk.TextBuffer.set_text() takes exactly 3 arguments (2 given)

That's quite strange. The `Gtk.TextBuffer.set_text` prototype [has not been changed recently](https://gitlab.gnome.org/GNOME/pygobject/-/blame/master/gi/overrides/Gtk.py#L758).

Which `python3-gobject*` packages do you have on your sysem? Can you provide the full output of the following commands?

```
rpm -qa | grep python3-gobject
rpm -qa | grep gtk3
python3 -c "import gi; gi.require_version('Gtk', '3.0'); from gi.repository import Gtk; print(Gtk.TextBuffer); help(Gtk.TextBuffer.set_text)" | cat
```

> What version of Gtk should I be using?

For the record my Gtk version is 3.24.1 and it's working fine so I doubt that the problem is in the Gtk version.

ChristopherRabotin, 2020-02-24:
Thanks for the quick reply. Here is the output you requested:

```
 chris@localhost  ~  rpm -qa | grep python3-gobject
python3-gobject-3.34.0-3.1.x86_64
python3-gobject-cairo-3.34.0-3.1.x86_64
 chris@localhost  ~  rpm -qa | grep gtk3
PackageKit-gtk3-module-1.1.12-16.3.x86_64
gtk3-branding-openSUSE-15.0-1.16.noarch
libcanberra-gtk3-module-0.30-15.8.x86_64
gtk3-metatheme-breeze-5.18.0-1.1.noarch
gtk3-data-3.24.13+0-3.3.noarch
gtk3-immodule-thai-3.24.13+0-3.3.x86_64
unity-gtk3-module-0.0.0+bzr20171202-1.8.x86_64
gtk3-immodule-vietnamese-3.24.13+0-3.3.x86_64
gtk3-schema-3.24.13+0-3.3.noarch
gtk3-immodule-amharic-3.24.13+0-3.3.x86_64
libqt5-qtbase-platformtheme-gtk3-5.14.1-1.3.x86_64
gtk3-immodule-tigrigna-3.24.13+0-3.3.x86_64
gtk3-immodule-inuktitut-3.24.13+0-3.3.x86_64
libcanberra-gtk3-0-0.30-15.8.x86_64
gtk3-metatheme-adwaita-3.28-1.8.noarch
kde-gtk-config5-gtk3-5.18.0-1.1.x86_64
libunity-gtk3-parser0-0.0.0+bzr20171202-1.8.x86_64
gtk3-tools-3.24.13+0-3.3.x86_64
 chris@localhost  ~  python3 -c "import gi; gi.require_version('Gtk', '3.0'); from gi.repository import Gtk; print(Gtk.TextBuffer); help(Gtk.TextBuffer.set_text)" | cat
<class 'gi.repository.Gtk.TextBuffer'>
Help on FunctionInfo in module gi.repository.Gtk object:

set_text = class FunctionInfo(CallableInfo)
 |  Method resolution order:
 |      FunctionInfo
 |      CallableInfo
 |      BaseInfo
 |      builtins.object
 |  
 |  Methods defined here:
 |  
 |  get_flags(...)
 |  
 |  get_property(...)
 |  
 |  get_symbol(...)
 |  
 |  get_vfunc(...)
 |  
 |  is_constructor(...)
 |  
 |  is_method(...)
 |  
 |  ----------------------------------------------------------------------
 |  Methods inherited from CallableInfo:
 |  
 |  __call__(self, /, *args, **kwargs)
 |      Call self as a function.
 |  
 |  can_throw_gerror(...)
 |  
 |  get_arguments(...)
 |  
 |  get_caller_owns(...)
 |  
 |  get_return_attribute(...)
 |  
 |  get_return_type(...)
 |  
 |  invoke(...)
 |  
 |  may_return_null(...)
 |  
 |  skip_return(...)
 |  
 |  ----------------------------------------------------------------------
 |  Methods inherited from BaseInfo:
 |  
 |  __eq__(self, value, /)
 |      Return self==value.
 |  
 |  __ge__(self, value, /)
 |      Return self>=value.
 |  
 |  __getattribute__(self, name, /)
 |      Return getattr(self, name).
 |  
 |  __gt__(self, value, /)
 |      Return self>value.
 |  
 |  __le__(self, value, /)
 |      Return self<=value.
 |  
 |  __lt__(self, value, /)
 |      Return self<value.
 |  
 |  __ne__(self, value, /)
 |      Return self!=value.
 |  
 |  __repr__(self, /)
 |      Return repr(self).
 |  
 |  equal(...)
 |  
 |  get_attribute(...)
 |  
 |  get_container(...)
 |  
 |  get_name(...)
 |  
 |  get_name_unescaped(...)
 |  
 |  get_namespace(...)
 |  
 |  get_type(...)
 |  
 |  is_deprecated(...)
 |  
 |  ----------------------------------------------------------------------
 |  Data and other attributes inherited from BaseInfo:
 |  
 |  __hash__ = None

 chris@localhost  ~ 
```

velichkov, 2020-02-25:
> rpm -qa | grep python3-gobject
> python3-gobject-3.34.0-3.1.x86_64
> python3-gobject-cairo-3.34.0-3.1.x86_64

It turns out in openSUSE the [Gtk.py](https://gitlab.gnome.org/GNOME/pygobject/-/blame/master/gi/overrides/Gtk.py#L758) file is provided by the `python3-gobject-Gdk` package but this package is not installed on your system.
```
$ rpm -qf /usr/lib64/python3.7/site-packages/gi/overrides/Gtk.py
python3-gobject-Gdk-3.34.0-3.1.x86_64
```
Try installing it and if it still fails the same way provide again the output of
```
python3 -c "import gi; gi.require_version('Gtk', '3.0'); from gi.repository import Gtk; print(Gtk.TextBuffer); help(Gtk.TextBuffer.set_text)" | cat
```
To me this looks like a packaging bug, the gnuradio package should depends on `python3-gobject-Gdk` and this package should be installed automatically when installing gnuradio. So if this helps please consider [reporting it to openSUSE](https://en.opensuse.org/openSUSE:Submitting_bug_reports) and post here a link to your bug report for reference.

dkozel, 2020-03-12:
Hello @ChristopherRabotin. Can you please let us know if installing `python3-gobject-Gdk` fixed your build? Since this is a most likely a packaging problem that we can't directly fix from the code here, I'm going to close the issue. If you don't think that's correct there's no problem re-opening it!

closed 2020-03-12 by dkozel