gnuradio/gnuradio #3245: impossible to launch gnuradio on ubuntu 18.04 LTS

dragonlost, 2020-03-04:
Hello.

I installed a new machine under Ubuntu 18.04LTS.
I only did a Sudo apt update and upgrade before launching the readme procedure:
- sudo apt install git python3 python3-numpy python3-pip
- sudo pip3 install --upgrade git + https: //github.com/gnuradio/pybombs.git
- sudo reboot
- auto-config pybombs
- add-defaults recipes pybombs
- pybombs prefix init ~ / gnuradio -R gnuradio-default

Then:
pybombs run gnuradio-companion

I get the following message:
In the terminal:
[INFO] Prefix Python version is: 3.6.9
[INFO] PyBOMBS Version 2.3.4a0
Gtk-Message: 11: 39: 17.041: GtkDialog mapped without a transient parent. This is discouraged.

in a window:
cannot import gnuradio.
Is the model path environment variable set correctly?
  All OS: PYTHONPATH

Is the library path environment variable set correctly?
Linux: LD_LIBRARY_PATH
Windows: PATH
MacOSX: DYLD_LIBRARY_PATH

(No module named 'gnuardio')

velichkov, 2020-03-04:
Hi @dragonlost,

See https://github.com/gnuradio/pybombs/issues/553 and adjust `PYTHONPATH` in your `setup_env.sh` script.

dragonlost, 2020-03-04:
thank you. For those who have the same problem as me: 
replace  : `~/gnuradio/lib/python3.6/dist-packages`
to `~/gnuradio/lib/python3/dist-packages` in `setup_env.sh`

closed 2020-03-04 by dragonlost