gnuradio/gnuradio #3250: Error when I try to use Embedded Python Block and to return lists of bits from it

All3xJ, 2020-03-06:
[image]
[image]

As you can see, this "Sequences_vectors" python embedded block that I created, has a list of bit sequences and it uses a for cycle to select each sequence and then return them to the next block (Repeat block) to go on in the flowgraph.
I did this to simulate a "Vector Source" block(that takes a _single_ list of bits), but that takes more than a single sequence of bits (in my case, as you can see in the code, there are three sequences).

I'd like to transmit each of these three sequences of bits. How can I do?

If I start the graph, it reports this errors:

```
Generating: '/Users/user/top_block.py'

Executing: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/user/top_block.py

handler caught exception: in method 'block_gw_message_type_work_args_return_value_set', argument 2 of type 'int'
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/gr/gateway.py", line 55, in eval
    try: self._callback()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/gr/gateway.py", line 225, in __gr_block_handle
    ) for i in range(noutputs)],
TypeError: in method 'block_gw_message_type_work_args_return_value_set', argument 2 of type 'int'
thread[thread-per-block[1]: <block Sequences_vectors (2)>]: SWIG director method error. Error detected when calling 'feval_ll.eval'
```

But even if it started without errors, I guess that the python embedded block wouldn't behave as expected. In fact I think it would return one list and stops, since the "return" only works one time and not more than one (so the other two sequences can't be "returned"). So I think my code wouldn't work.

velichkov, 2020-03-08:
Hi @All3xJ,

Please never attach screenshots of your code but rather copy-paste it as text and use [markdown] for formatting.

> and to return lists of bits from it

You are not suppose to return lists from the `work` function but the number of elements you put in the `output_items` buffers. So you need to copy your lists to the `output_items` and return the length.

See also [Blocks Coding Guide](https://wiki.gnuradio.org/index.php/BlocksCodingGuide#The_work_function) and [Embedded PythonBlock](https://wiki.gnuradio.org/index.php/Embedded_Python_Block) documents.

dkozel, 2020-03-12:
@All3xJ this type of application debugging question is probably best on the chat channel or mailing list.
https://wiki.gnuradio.org/index.php/MailingLists

Hopefully the solution in https://github.com/gnuradio/gnuradio/issues/3251 will work for you.

closed 2020-03-12 by dkozel