gnuradio/gnuradio #3251: GNURadio doesn't let me work with more than one sequence of bits.

All3xJ, 2020-03-06:
[image]

[image]

[image]

I am using "**Vector Source**" block. As you know, it's used with a **list of ints**. Of course, if I try to use more than a list, so if I use a list of lists, it gives me error:

```
Traceback (most recent call last):
  File "/Users/user/top_block.py", line 160, in <module>
    main()
  File "/Users/user/top_block.py", line 148, in main
    tb = top_block_cls()
  File "/Users/user/top_block.py", line 90, in __init__
    self.blocks_vector_source_x_0 = blocks.vector_source_c([x for x in sequence], True, 1, [])
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/blocks/blocks_swig1.py", line 1593, in make
    return _blocks_swig1.vector_source_c_make(*args, **kwargs)
TypeError: in method 'vector_source_c_make', argument 1 of type 'std::vector< gr_complex,std::allocator< gr_complex > > const &'

>>> Done (return code 1)

```

I also tried with:
[image]
But it doesn't work neither (same error of before)

Obviously if I write `sequence[0]` to select _only the first_ sequence, it works, and no errors:

[image]

**So my question is, how can I use more lists with "Vector Source" block?**

velichkov, 2020-03-08:
Hi @All3xJ,

> [x for x in sequence] 

Try converting the list of lists to a list (e.g. flattening the list)
```
[item for sublist in sequence for item in sublist]
```

dkozel, 2020-03-12:
I'm not sure if this is a useful feature to add. @All3xJ could you give some detail about why a list of lists was the natural answer here? I think Vasil's advice is the short term, and maybe long term, answer.

`[sample for sublist in [[0,1],[1,0],[1,1],[0,0]] for sample in sublist]`

mormj, 2020-04-09:
Closing, as list of lists is not the desired functionality of the vector source, and would be inconsistent with the rest of the api

closed 2020-04-09 by mormj