gnuradio/gnuradio #3252: QT GUI component requires gr-analog but cmake doesn't require it

doronbehar, 2020-03-10:
On NixOS we'd like to provide complete support for enabling and disabling features / components in Gnuradio 3.8.0.0. While testing with some of them turned on & off, I encountered the following error when `gr-analog` was turned off and the QT GUI was turned on:

```
[ 94%] Building CXX object gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/display_qt.cc.o
In file included from /build/source/gr-qtgui/examples/c++/display_qt.cc:22:
/build/source/gr-qtgui/examples/c++/display_qt.h:24:10: fatal error: gnuradio/analog/noise_source.h: No such file or directory
   24 | #include <gnuradio/analog/noise_source.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/build.make:67: gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/display_qt.cc.
] Error 1
make[1]: *** [CMakeFiles/Makefile2:5464: gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Enabling `gr-analog` fixes this.

From what I saw, Cmake refused to even start building if a certain component is required for another. But in this case, the GUI component didn't require `gr-analog` hence this bug report.

dkozel, 2020-03-12:
Thanks for the report. That should get added as a dependency in CMake. I think it is reasonable for the GUI to rely on gr-analog as it's unlike that anyone will install GNU Radio without gr-analog.

mormj, 2022-09-09:
I believe this is duplicate/related to #2877, which appears to have been fixed by https://github.com/gnuradio/gnuradio/commit/7470a7a3771dd90defb826b464dfe62977cb1eb6

Can you please verify?

doronbehar, 2022-09-10:
I just tried building GR with the following features:

```
-- ######################################################
-- # Gnuradio enabled components
-- ######################################################
--   * testing-support
--   * python-support
--   * post-install
--   * man-pages
--   * gnuradio-runtime
--   * common-precompiled-headers
--   * gnuradio-companion
--   * JSON/YAML config blocks
--   * gr-blocks
--   * gr-fec
--   * gr-fft
--   * gr-filter
--   * gr-audio
--   * * alsa
--   * * oss
--   * * jack
--   * gr-pdu
--   * gr-iio
--   * * libad9361
--   * gr-qtgui
--   * gr-video-sdl
--   * gr-zeromq
--   * gr-network
--   * gr-soapy
--
-- ######################################################
-- # Gnuradio disabled components
-- ######################################################
--   * doxygen
--   * gr-ctrlport
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-channels
--   * gr-trellis
--   * gr-uhd
--   * gr-uhd UHD 4.0 RFNoC
--   * gr-utils
--   * gr_modtool
--   * gr_blocktool
--   * gr-vocoder
--   * gr-wavelet
```

And it succeeded. Thanks for fixing the issue.

closed 2022-09-10 by doronbehar