gnuradio/gnuradio #3254: [Question] Opus codec block

Radiomix2000, 2020-03-11:
Hello!
Any chance to see the block with opus codec in GR 3.8.x? It was announced before release of v3.8, but there is no opus codec after release..

dkozel, 2020-03-12:
Hi @Radiomix2000 The Opus blocks didn't make it into the release, none were ready for merging as far as I know. @ast was interested and working with Opus in GNU Radio I believe.

Is this something you'd be interested in working on?

ast, 2020-03-12:
I wrote an Opus block that output a tagged stream (opus packets are variable length). Not sure how that would integrated into gr-vocoder? It's straight forward thing except Opus has a lot of different settings you can tweak.

Radiomix2000, 2020-03-15:
> Is this something you'd be interested in working on?
Considering my poor programming experience (novice) I can help as much, as my knowledge allows. I'm still learning.

> Not sure how that would integrated into gr-vocoder?
Not necessary, it could be an independent OOT block.

marcusmueller, 2020-04-12:
@ast but libopus automatically demuxes these different-size packets if you just put them in a stream consecutively?

ast, 2020-04-12:
But you want an UDP sink after the opus block that sends the packet as soon as it's ready and not wait until it has 1k bytes or similar. So you can't output a stream and you can't output fixed length vector. A tagged stream makes most sense to me.

Radiomix2000, 2021-02-07:
This repo https://github.com/qradiolink/qradiolink/ is using gnuradio graph and opus codec in QPSK transmission. As I know from author and if I don't mistake, he uses opus in fixed frame length mode. So no VBR, only CBR, set the frame length, and we have fixed packets. Did I understand it right?
9400 bps in above mentioned software sound really nice and silky. Very natural voice sound on decoder side.
Sad, that nobody updates codecs in GR. There are only old and outdated codecs, except codec2, included with GR.

PR #5829 (2022-05-12): blocks: Add support for Ogg Opus if libsndfile is >= 1.0.29
## Description
libsndfile added support for Ogg Opus in version 1.0.29. This pull request makes it accessible through the Wav File Sink block.

## Related Issue
* Fixes #3254.

## Which blocks/areas does this affect?
* Wav File Sink

## Testing Done
I verified that both Opus and Vorbis output work after this change, and that an exception is raised when Opus is not available.

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to h [... cut]
closed 2022-05-12 by mormj