gnuradio/gnuradio #3256: Python Module ID Missing 3.8.0.0

maxwell-maurice, 2020-03-11:
Hello, I am running 

OS: Manjaro Linux x86_64 
Kernel: 5.4.23-1-MANJARO 
GNU-Companion: 3.8.0.0 (python 3.8.1)

and it looks like the ID is missing from the python module block. Running 3.7.1 in Ubuntu I see the ID and can work from there. Any way to find out the ID of the block? 
[image]

dkozel, 2020-03-12:
Hi @maxwell-maurice thanks for the report. Looks like it was removed in 3.8.0.0. The default value has not changed and you could see it by looking at the file name of the generated `.py` file after running or generating the flowgraph.

@mormj and I just has a quick thought last night about the ID of the `Python Block` block and making it user specifiable. It looks like this is actually a regression so probably pretty easy to fix.

mormj, 2020-03-12:
At least on master the block ids are exposed - I'm not seeing them on maint-3.8.  Also on master there is a feature to show all block ids which can be useful - this should be ported back to maint-3.8 as well.  I'll put together a few pull requests this afternoon to address

PR #3257 (2020-03-13): grc: backport show_id related commits to maint-3.8
Closes: #3256 

This addresses the situation that blocks explicitly need a show_id flag in 3.8.  Several blocks whose block ids need to be given to other variables or blocks need this populated, and had been updated on master. 

This also backport the feature to 3.8 where a toggle is given to show ids for all blocks.
closed 2020-03-26 by dkozel