gnuradio/gnuradio #3260: grc: embedded python objects fail to load because of missing cpp_templates

mormj, 2020-03-13:
`
AttributeError: 'EPyBlock' object has no attribute 'cpp_templates'
`
This is with latest master edaa88fbc889a3d09fd374c3eb49cd46174b049f

Reproduce by opening a blank flowgraph in grc, and adding "Python Block" or "Python Module"

Going to see if there is an easy fix, but wanted to go ahead and post the issue

mormj, 2020-03-13:
As the classes that derive from block load and get passed to this decorator, the flags accumulate in all the classes

dummy - {}
epy_block - {show_id}
virtual_source - {show_id, cpp}
...

so by the time virtual source is added, the full list build_ins in each class has all the flags

```
def register_build_in(cls):
    cls.loaded_from = '(build-in)'
    build_ins[cls.key] = cls
    return cls
```

dkozel, 2020-03-13:
Do you know what changed caused these blocks to stop working?

mormj, 2020-03-13:
Looks like some of the recent cpp_template changes.  The underlying issue that these blocks have flags that they shouldn't was probably always there, but the recent change to access the cpp_templates member of a block is new, and that's where the exception was occuring

PR #3261 (2020-03-26): grc: workaround to allow epy blocks/modules to load
The underlying issue is that the built in blocks all end up getting the
'cpp' flag, which is not valid for the embedded python blocks

To fix this some work needs to be done in the "Flags" class, as all the
flags added to classes that get accumulated in the build_ins list get
added to all the classes

See register_build_in decorator

Fixes #3260 for now, though a more thorough solution is needed going forward
closed 2020-03-26 by michaelld