gnuradio/gnuradio #3267: Does installing gnuradio require using Debian "sid" unstable repos?

Allasso, 2020-03-17:
I get "unmet dependencies" error when attempting to install on Ubuntu 18.04 using apt.  After tracing back, I find it is attempting to install packages (eg, libvolk2.2) which are (apparently) only hosted on sid (unstable) repos.  Does installing gnuradio require I add unstable repos to my sources?

I just want to check for sure if this is the case, and make sure I'm not missing something simple.

Complete details are described in this superuser post:
https://superuser.com/questions/1533167/unmet-dependencies-error-when-trying-to-install-gnuradio-on-ubuntu-18-04

Allasso, 2020-03-17:
Would like to mention also that this is a fresh install of 18.04, and AFAICT the auto software updater has everything up to date as of yesterday.

mormj, 2020-03-17:
It looks like I did not update the volk dependencies in launchpad.  The files are copying now

Note that this is only an issue for master branch builds - @Allasso are you trying to use the latest master branch, or the latest 3.8 release?  If the latter, then you probably don't want to add all 3 ppas

Allasso, 2020-03-17:
Thanks.  I removed the ppas and ran it again, and everything seemed to work fine.  The only weird thing with the install is at the end:

Failed to parse '/usr/share/mime/packages/gnuradio.xml'

But I am guessing that is an apt thing and probably benign.

I don't need to use the latest master.  Thanks for updating the volk dependencies.

closed 2020-03-17 by Allasso