gnuradio/gnuradio #3268: soft_dec_lut_gen.py update needed, float being used as integer index

BCribbs, 2020-03-18:
in the calls to linspace, npts is computed using floats and then attempting to be used as an integer.  The change is to wrap the call in int() to ensure its an integer, which allows it to 
pass qa_constellation tests.

Code change is:
npts = int(2.0**prec)

This occurs in two places.

nickoe, 2020-04-01:
Mm, when I run the qa_constellation test on maint-3.8 the test passes. So I am not quite sure I follow you.

velichkov, 2020-04-01:
Hi @nickoe,

Yes, the test passes but the following 4 warnings are printed when you run the test manually or with `ctest -V`
```
$ ./build/gr-digital/python/digital/qa_constellation_test.sh
gr-digital/python/digital/soft_dec_lut_gen.py:114: DeprecationWarning: object of type <class 'float'> cannot be safely interpreted as an integer.
gr-digital/python/digital/soft_dec_lut_gen.py:115: DeprecationWarning: object of type <class 'float'> cannot be safely interpreted as an integer.
gr-digital/python/digital/soft_dec_lut_gen.py:78: DeprecationWarning: object of type <class 'float'> cannot be safely interpreted as an integer.
gr-digital/python/digital/soft_dec_lut_gen.py:79: DeprecationWarning: object of type <class 'float'> cannot be safely interpreted as an integer.
```
and according to the [numpy's documentation](https://numpy.org/devdocs/reference/generated/numpy.linspace.html) the 3rd parameter should be `int`.

nickoe, 2020-04-01:
@ah, ok, I see that with python3 indeed.

BCribbs, 2020-04-01:
Fails on 3.9,  by chance are still running Python 2.7 or 3.6. 

qa_ fails consistently under 3.9 straight out of git clone and Python 3.6

npts = int(2.0**prec)

Fixes the problem, the int() does the correct casting.

Brown

PR #3351 (2020-04-12): gr-digital/python : Updated soft_dec_lut_gen.py to avoid depreciation warnings.
Fixes: #3268 
In soft_dec_lut_gen.py, npts is a float and is used as third parameter of numpy.linspace which should be int according to [numpy's documentation](https://numpy.org/devdocs/reference/generated/numpy.linspace.html). This PR removes the warnings.
closed 2020-04-12 by mbr0wn