gnuradio/gnuradio #3271: gnuradio companion: “sources” branch does not appear in blocks menu in Ubuntu 18.04

Allasso, 2020-03-20:
Ubuntu 18.04.  

I have done the following:

```
sudo apt install gnuradio
sudo apt install gr-osmosdr
sudo apt install hackrf
sudo apt install libhackrf-dev
```

Following the gnuradio tutorial here:

https://greatscottgadgets.com/sdr/1/

After launching gnuradio companion, I am unable to add a source as "sources" does not appear in the blocks menu.

Allasso, 2020-03-20:
Additional info, I am trying to use HackRF One device.  All appropriate lights are lit.  Tried resetting the device.

drmpeg, 2020-03-20:
Take a look in the "(no module specified)" category.

Allasso, 2020-03-20:
Thanks, there it is.  I set up companion on MacOS and it was in the Core list, so was looking for it there.

closed 2020-03-20 by Allasso