gnuradio/gnuradio #3273: gr-gsm error

tinkerermaruthu, 2020-03-21:
sudo pybombs install gr-gsm
[sudo] password for maruthu: 
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 3.7.6
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gr-gsm
   |
   +- gr-osmosdr
   |  |
   |  +- gr-iqbal
   |  |  |
   |  |  \- gnuradio
   |  |
   |  +- soapysdr
   |  |
   |  \- gnuradio
   |
   \- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.PackageManager - WARNING - A source build for package gnuradio was requested, but binary install was found!
Install gnuradio from source despite binary install available Y/[N]? y
PyBOMBS.Packager.source - WARNING - Build dir already exists: /home/maruthu/src/gnuradio/build
Configuring: (100%) [==================================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring testing-support support...
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- Extracting version information from git describe...
-- Compiler Version: cc (Debian 9.2.1-30) 9.2.1 20200224
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python3.6
-- Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is exact version "3.6")
Traceback (most recent call last):
  File "<string>", line 2, in <module>
ImportError: cannot import name 'sysconfig'
CMake Error at cmake/Modules/GrPython.cmake:117 (file):
  file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
  exactly three arguments.
Call Stack (most recent call first):
  CMakeLists.txt:362 (include)

Traceback (most recent call last):
  File "<string>", line 2, in <module>
ImportError: cannot import name 'sysconfig'
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = FALSE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled python-support but configuration checked failed
Call Stack (most recent call first):
  CMakeLists.txt:375 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/maruthu/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/maruthu/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.

velichkov, 2020-03-21:
Hi @tinkerermaruthu,

> PyBOMBS - INFO - PyBOMBS Version 2.3.3

Your `pybombs` version is pretty old. Better update to [the latest version from github](https://github.com/gnuradio/pybombs/#using-python3)
```
sudo pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
```

> -- Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is exact version "3.6")

The latest version of pybombs [should install python3-dev package](https://github.com/gnuradio/pybombs/blob/4c8dc5b1285d2ff3c700f3be16d23591c3d4daba/pybombs/recipes/python.lwr#L28-L31) and resolve this problem

FFY00, 2020-03-22:
Well, PyBOMBS 2.3.3 is the latest release, maybe a new release should be tagged.

dkozel, 2020-03-26:
If this issue is present with the most recent pybombs code then can you please reopen an issue on the PyBOMBS issue tracker? Thanks!

Yes, tagging a release of PyBOMBS seems long overdue.

closed 2020-03-26 by dkozel