gnuradio/gnuradio #3279: Add -undefined dynamic_lookup linker flag back for macOS

iMichka, 2020-03-24:
Hi

Homebrew maintainer here.
We are trying to update gnuradio to version 3.8.0.0.
https://github.com/Homebrew/homebrew-core/pull/49733

It looks like the python `-undefined dynamic_lookup` linker flag has been removed in the last release. See https://github.com/gnuradio/gnuradio/commit/ab2fb35677e38a384df3f9503d1f45f64bbc0374#diff-45856d0ec64e3d28b38f044024648892L163

Could this be added back?

michaelld, 2020-03-24:
Hi Michka - It -could- ... but, that doesn't seem necessary in general. Maybe explain what's going on & why you think it -should- be added back? I mean, you can always apply a patch in brew that reverts that commit ... yes?

iMichka, 2020-03-24:
Hi.

The thing is that we will build `gnuradio` against brewed Python 3.8. This means that we will link against that Python.

If later on the user has another Python 3 in his path, running `gnuradio` will fail with the `Abort Trap 6` error message. (I did not test this but this issue has happened so many times that we require all libraries that link against Python to use the `-undefined dynamic_lookup` flag).

On Linux you do not need this flag, as this is the default behaviour of the compiler. On mac you need to explicitly ask for the dynamic lookup.

I tried to set the flag manually by prepending it to the LDFLAGS, but it did not help. I also tried to hack around the cmake files but was not able to find the right place to add the flag (or I found it but something went wrong and it did not work).

michaelld, 2020-03-25:
Interesting. As you note, that "feature" was removed in the massive "modernize cmake" commit ... and, really, I don't think it should have been. I thought I'd debugged that change & it worked for me ... but now I wonder. Let me do some local testing to see what's going on.

iMichka, 2020-03-25:
Just a small hint on how to debug this (if you need a way to check if the fix is working):

```
oTool -L /usr/local/Cellar/gnuradio/3.8.0.0/lib/python3.7/site-packages/gnuradio/audio/_audio_swig.so 
/usr/local/Cellar/gnuradio/3.8.0.0/lib/python3.7/site-packages/gnuradio/audio/_audio_swig.so:
	/usr/local/Cellar/gnuradio/3.8.0.0/lib/libgnuradio-audio.3.8.0.dylib (compatibility version 3.8.0, current version 3.8.0)
	/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python (compatibility version 3.7.0, current version 3.7.0)
	/usr/local/Cellar/gnuradio/3.8.0.0/lib/libgnuradio-runtime.3.8.0.dylib (compatibility version 3.8.0, current version 3.8.0)
	/usr/local/Cellar/gnuradio/3.8.0.0/lib/libgnuradio-pmt.3.8.0.dylib (compatibility version 3.8.0, current version 3.8.0)
	/usr/local/Cellar/gnuradio/3.8.0.0/lib/libvolk.2.0.dylib (compatibility version 2.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)
	/usr/local/opt/boost/lib/libboost_program_options.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_system.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_regex.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_thread-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/log4cpp/lib/liblog4cpp.5.dylib (compatibility version 6.0.0, current version 6.6.0)
	/usr/local/opt/gmp/lib/libgmpxx.4.dylib (compatibility version 11.0.0, current version 11.0.0)
	/usr/local/opt/gmp/lib/libgmp.10.dylib (compatibility version 15.0.0, current version 15.0.0)
	/usr/local/opt/jack/lib/libjack.0.dylib (compatibility version 1.0.0, current version 1.28.0)
	/System/Library/Frameworks/AudioUnit.framework/Versions/A/AudioUnit (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreAudio.framework/Versions/A/CoreAudio (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/AudioToolbox.framework/Versions/A/AudioToolbox (compatibility version 1.0.0, current version 1000.0.0)
	/System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 162.0.0)
	/usr/local/opt/portaudio/lib/libportaudio.2.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 800.7.0)

As you can see, `/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python` is present, which is brewed Python 3. Ideally that line should not be there.

michaelld, 2020-03-25:
Yup. That's what I'd guess I'll see too. Still updating. Going to take me a few days to get my installs updated, verify the issue, & get a PR in the GR queue ...

Bo98, 2020-04-03:
For comparison, you can see how CMake 3.15's `Python::Module` target handles it.

iMichka, 2020-04-26:
Ping @michaelld, sorry to bother you. Did you have time to look at this? Maybe you have some hints on how to handle this. I could try to help (though I already tried to add the flag by modifying the cmake code, and failed).

michaelld, 2020-04-26:
@iMichka I'm trying to get a PR finalized in MacPorts to update all of the GR ports. Once that's done, I'll test this issue out. It will take a few more days -- maybe a week -- but I'll get there. The PR will update GR to 3.8.1 release, which will make testing this issue really easy.

iMichka, 2020-05-22:
@michaelld sorry to ping you again :( Had you time to look at this?

PR #3674 (2020-08-03): cmake: Add -undefined dynamic_lookup linker flag back for macOS
Closes #3279
closed 2020-08-03 by michaelld