gnuradio/gnuradio #3281: qtgui: matplotlib dependency

mormj, 2020-03-27:
When `__init.py__` is loaded for gr-qtgui, the following code in distanceradar.py causes a failure because matplotlib is missing.  This needs to be reworked as an optional dependency

```
import sys
from PyQt5 import QtWidgets
import numpy as np
import matplotlib.pyplot as plt
try:
    from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
    from matplotlib.figure import Figure
except:
    gr.log.error("Unable to import matplotlib.  Please install matplotlib first "
                 "(e.g., via pip/pip3/dpkg/MacPorts).")
    sys.exit(1)
    
from gnuradio import gr
import pmt
```

drmpeg, 2020-03-29:
I noticed this too. But how are the CI tests still passing?

mormj, 2020-03-31:
That's a good question - it would only get hit in the qtgui CI test, which does import qtgui obviously.  So why it passes is a mystery to me.  I'll take a look into it.

sidkapoor97, 2020-04-08:
I  removed the exit condition and made the import matplotlib as optional. Although couldn't figure out as to why tests are passing

mormj, 2020-04-08:
@sidkapoor97 - thank you for the condition. The problem with having no exit condition is that if matplotlib is not installed, the __init__ import will hit the exception on loading DistanceRadar.py and the whole module will fail to load

Try putting the check in __init__.py so that the load of matplotlib dependent blocks is conditional, e.g.
```
try:
    import matplotlib.pyplot as plt
    from .distanceradar import DistanceRadar
    from .azelplot import AzElPlot
except ModuleNotFoundError:
    from gnuradio import gr
    gr.log.warn("Matplotlib is a required dependency to use DistanceRadar and AzElPlot."
                "  Please install matplotlib to use these blocks (https://matplotlib.org/)")
```

Still not sure why the qa tests are failing in CI - they fail on my system when I uninstall matplotlib and I don't see matplotlib installed in the buildbot docker images

drmpeg, 2020-04-08:
I figured it out. The qtgui test is excluded in the CI command line.

ctest --output-on-failure --timeout 120 -E qtgui

sidkapoor97, 2020-04-08:
I have added the changes as requested by @mormj , also I am sorry for any stupid mistakes, I am fairly new and trying to contribute as much as possible

mormj, 2020-04-08:
No stupid mistakes - the contributions are much appreciated!  Glad you are getting involved

mormj, 2020-04-09:
closed with #3322

closed 2020-04-09 by mormj