gnuradio/gnuradio #3282: examples: test_msg_hier.grc doesn't load

mormj, 2020-03-27:
I think a similar issue has been fixed in other flowgraphs, but documenting here:

[image]

```
Loading: "/share/gnuradio/grdev/src/gnuradio/gr-blocks/examples/msg_passing/hier/test_msg_hier_topblock.grc"
>>> Loading: /share/gnuradio/grdev/src/gnuradio/gr-blocks/examples/msg_passing/hier/test_msg_hier.grc
>>> Error: Connection between blocks_message_strobe_0(strobe) and pad_sink_0(in0) could not be made.
	sink key 'in0' not in sink block keys
>>> Load Error: /share/gnuradio/grdev/src/gnuradio/gr-blocks/examples/msg_passing/hier/test_msg_hier.grc: Flowgraph invalid
```

nickoe, 2020-03-27:
What branch is this on?

mormj, 2020-03-27:
> What branch is this on?

master

nickoe, 2020-03-27:
It looks like this on maint-3.8 (a65d1d1c236f95e4a106db6431ff4d51021dc29c)

```
>>> Error: Connection between blocks_message_strobe_0(strobe) and pad_sink_0(in0) could not be made.
	sink key 'in0' not in sink block keys
Traceback (most recent call last):
  File "/tmp/gnuradio/lib/python3.8/site-packages/gnuradio/grc/core/FlowGraph.py", line 418, in import_data
    sink_port = verify_and_get_port(snk_port_id, sink_block, 'sink')
  File "/tmp/gnuradio/lib/python3.8/site-packages/gnuradio/grc/core/FlowGraph.py", line 399, in verify_and_get_port
    raise LookupError('%s key %r not in %s block keys' % (dir, key, dir))
LookupError: sink key 'in0' not in sink block keys
```

nickoe, 2020-03-27:
Adding the connections and diffing what is saved gives:
```
...
@@ -118,16 +136,18 @@ blocks:
     type: message
     vlen: '1'
   states:
+    bus_sink: false
+    bus_source: false
+    bus_structure: null
     coordinate: [120, 148.0]
     rotation: 0
     state: enabled
 
 connections:
-- [blocks_message_strobe_0, strobe, pad_sink_0, in0]
-- [blocks_message_strobe_0_1, strobe, pad_sink_0, in0]
-- [blocks_message_strobe_0_1, strobe, pad_sink_1, in0]
-- [pad_source_0, out0, blocks_message_strobe_0, set_msg]
-- [pad_source_1, out0, blocks_message_strobe_0, set_msg]
+- [blocks_message_strobe_0, strobe, pad_sink_0, in]
+- [blocks_message_strobe_0_1, strobe, pad_sink_1, in]
+- [pad_source_0, out, blocks_message_strobe_0, set_msg]
+- [pad_source_1, out, blocks_message_strobe_0_1, set_msg]
```

nickoe, 2020-03-27:
If the "Num Streams" are set to 1 the single input port is called "in", if there are multiple they are suffixed with a number, as in "in0", "in1"...

nickoe, 2020-03-27:
This commit specifically mentions in0. https://github.com/gnuradio/gnuradio/pull/2329/commits/999da062db09ade129461671127973bd379830a8

@jdemel Do you happen to know what is the "right" thing to do here?

mormj, 2020-03-28:
> It looks like this on maint-3.8 ([a65d1d1](https://github.com/gnuradio/gnuradio/commit

It must be on both - probably never got updated in the conversion from xml to yaml

Did redoing the connections seem to fix the flowgraph?  I think the commit you referenced could be just a problem with the xml to yaml converter, but manually fixing makes the problem go away.

nickoe, 2020-03-28:
Yes, it seems to fix the loading of the grc. I am not sure how to test it really, since it is a hier block.

PR #3289 (2020-03-30): Fixup some grc example files in; blocks and digital
Fixes #3282 and partly fixes unresolved notes in #2285

Backport of https://github.com/gnuradio/gnuradio/pull/3286 maint-3.8 fix onto master.

The docs change is not needed on master as it seems that has been fixed and moved to gr-analog.
commit 2e21f040 (2020-03-30): blocks: Re-convert GRC examples

This is made by opening the grc file in gnuradio-companion and saving a
copy as on the same file.

Fixes #3282
commit 555cf4ad (2020-03-30): blocks: Re-convert GRC examples

This is made by opening the grc file in gnuradio-companion and saving a
copy as on the same file.

Fixes #3282
closed 2020-03-30 by mbr0wn