gnuradio/gnuradio #3302: GRC: make middle mouse click on tab close flowgraphs

mormj, 2020-04-02:
Just like what happens when you click on a browser tab with the middle mouse button (wheel)

sidkapoor97, 2020-04-08:
I believe this will be edited in the grc/gui/actions.py [1] under Old Actions 
also references #3145
[1] https://github.com/gnuradio/gnuradio/blob/master/grc/gui/Actions.py

mormj, 2020-04-09:
@sidkapoor97 - are you interested in addressing this feature?

sidkapoor97, 2020-04-09:
Yes I wanted to @mormj I tried going through related issues and PRs

sidkapoor97, 2020-04-14:
@mormj I tried making changes to gui/Actions.py and gui/Applications.py, I needed help with the 
`keypresses=["Middle Mouse button"]`

`FLOW_GRAPH_CLOSE_MIDDLE_MOUSE = actions.register("app.flowgraph.close",
    label='_Close using the middle mouse',
    tooltip='Close the current flow graph using Middle Mouse',
    icon_name='window-close',
    keypresses=[" "],
)`

These were the following links i referred [https://gitlab.gnome.org/GNOME/gtk/-/blob/3.24.14/gdk/keyname-table.h](url)
[https://developer.gnome.org/pygtk/stable/class-gdkevent.html#constructor-gdkevent](url) - under `gtk.gdk.BUTTON_PRESS` "button"

mormj, 2020-04-16:
@sidkapoor97 - I actually haven't looked into the implementation here, was just throwing it out as a feature enhancement.  Does your code as shown above work when added to GRC?

sidkapoor97, 2020-04-16:
@mormj I am not clear on how to test that feature, I tried editing the script and installing gnu radio from source but encountered some errors.  any help on testing would be really appreciated, once I test the feature i will open up a PR

dkozel, 2020-04-17:
@sidkapoor97 You should first solve all the build errors without making any changes to the script. When GNU Radio works normally, then make your changes. If there are errors with your changes then you will know where the error must be.

To test this feature, middle click on the tab of a flowgraph and it should close. If the flowgraph is unsaved then it should ask to save first, like if you close an unsaved flowgraph already without this feature.