gnuradio/gnuradio #3309: gr-filter: fir_filter kernels are not thread-safe

jacobgilbert-snl, 2020-04-04:
Because the FIR filter kernels store output data temporarily in member variables (see: https://github.com/gnuradio/gnuradio/blob/master/gr-filter/lib/fir_filter.cc#L130) use of this is not safe within multithreaded blocks. Simple fix is to use local variables.

This issue was observed with this OOT module which uses omp for multithreading: https://github.com/sandialabs/gr-fhss_utils/blob/master/lib/tagged_burst_to_pdu_impl.cc

jacobgilbert-snl, 2020-04-04:
Possibly redundant to https://github.com/gnuradio/gnuradio/pull/3062

mrjacobagilbert, 2020-04-05:
So not being an expert in volk, I'm not sure our solution of using local variables is the best option as they won't necessarily be VOLK aligned.

mrjacobagilbert, 2020-04-06:
Not really a GR problem, will address in OOT.

closed 2020-04-08 by jacobgilbert-snl