gnuradio/gnuradio #3312: GNURadio 3.8.0.0: Stream Mux is broken

surpscare, 2020-04-05:
I just installed GNURadio at Manjaro 19.0.2 Kyria, and tried to execute a flowgraph made with older one in the Windows build. So I got an error at line in generated script which is notorious for Stream Mux block. It says:

TypeError: make() takes 2 positional arguments but 6 were given
[image]

So 6 is an exact number of 5 Mux inputs + "gr.sizeof_gr_complex*1" value.
[image]

I tried to reproduce this in simple flowgraph (4-input Stream Mux block + 4 constant sources + decomplexor + audio sink) and same thing happens, now is "5 position arguments" respectively.
[image]
(Also, why there is so large top panel buttons? There are no screenshot in the Web with such layout... And why the blocks doesn't connect with click&drag? And why there's an error which gives 4/2=2.0 and says "Can not cast evaluated value '2.0' to type <class 'int'>" which leads to unability of math operations with variables in value field, as earlier? Too many questions to fit in one bugreport...)

dkozel, 2020-04-05:
Hi @surpscare Thanks for the issue report. As you note, there are a bunch of different issues/questions that you list. Can you please open new issues for them separately?

1) The large buttons must be a GTK theming issue in Manjaro, I'm not sure how to debug that. Can you bring it up with the Manjaro packager?
2) Connecting blocks using click and drag has never worked, it'd be a great feature to add
3) Python 3 changed the way integer division works compared to Python 2. 4/2 used to equal 2, now it is 2.0 so we have this frequent error. It would be nice if we could auto-detect this in GRC (maybe check if the number could be converted to an integer without a remainder?)

On the main issue, I can replicate the issue on the head of `maint-3.8`.
The generated Python code is
`blocks.stream_mux(gr.sizeof_gr_complex*1, 1,1,1,1)`
but it should be
`blocks.stream_mux(gr.sizeof_gr_complex*1, [1,1,1,1])`

To test this you can change lengths to `[40,5,2,220]` in your example.

surpscare, 2020-04-05:
Thanks! I just discovered that click+drag connection feature wasn't presented in older version (3.7.12.0), it was just a false memory caused because arrow is now appears when trying to click-and-drag, which wasn't earlier.
I got a new issue: version maint-3.8 had converted GRC project to new non-XML format, and now I have no option to convert it back to test with stable versions....

dkozel, 2020-04-05:
Yes, it's not good that GNU Radio overwrites the flowgraph file without warning!

Two quick bug fixes that I tried both didn't work to solve the Stream Mux issue. I'll have another look tomorrow but don't have a good answer yet for a PR.

surpscare, 2020-04-05:
Thanks! Square brackets helped.

I also tried to build version 3.7.14 by following Wiki instructions, but it failed at cmake generation step. PyBOMBS variant fails too with warning about "Python 2.7 is deprecated". Here's the error output files, I just don't know where to ask questions about building obsolete versions as long as they no longer maintained.
(However, square brackets helped!)
[CMakeError.log](https://github.com/gnuradio/gnuradio/files/4435263/CMakeError.log)

dkozel, 2020-04-08:
You can try asking in the IRC/Slack channel or on the mailing list. Posting the full output of cmake and make will help the most.

willcode, 2021-03-03:
Solutions:
1) change the default from `1, 1` to `[1, 1]`
2) interpret the input as the contents of a list

I think that (1) would be better so the list could also be specified in a variable, and so that it's clear whenever a list type is being used. Filter taps, for example, need to be specified as a list.

marcusmueller, 2024-01-08:
raw list entry does work now, but

closed 2024-01-08 by marcusmueller