gnuradio/gnuradio #3313: gr::block(2,1) filled with same inputs seems to be losing data

Math32-ui, 2020-04-06:
Hello all,

I am making a flowgraph with two independant file sources which are connected to their respective input filter and resampler. Respective outputs are connected to a single block, created with two input ports and one output.
In the general_work function, when I am trying to read the received data, I do not have the exact same values in my input vectors even if I select identical source files.

I read my input as follow : 

`general_work(int noutput_items __attribute__((unused)), gr_vector_int &ninput_items,
    gr_vector_const_void_star &input_items, gr_vector_void_star &output_items)

{

    gr::thread::scoped_lock l(d_setlock);
    const auto *in1 = reinterpret_cast<const gr_complex *>(input_items[0]);
    const auto *in2 = reinterpret_cast<const gr_complex *>(input_items[1]);

std::cout << in1[1].real() << "+i." << in1[1].imag() << std::endl;
std::cout << in2[1].real() << "+i." << in2[1].imag() << std::endl;`

My forecast function is as follow :

`forecast(int noutput_items,  gr_vector_int &ninput_items_required)
{
    if (noutput_items != 0)
        {
            ninput_items_required[0] = static_cast<int32_t>(trk_parameters.vector_length) * 2;
            ninput_items_required[1] = static_cast<int32_t>(trk_parameters.vector_length) * 2;
        }
}`

I work with an internal sampling frequency of 2MHz (original source file sampling frequency = 20MHz).

I can't figure if my issue is related to an high internal sampling frequency, or to a misconfiguration of my block (and of the global synchronization of the flowgraph). 

Is somebody have an idea concerning my issue?

Thanks !

nickoe, 2020-04-06:
@Math32-ui I have glanced over it three times now, but yet I can not find what version of gnuradio you are using.

Math32-ui, 2020-04-07:
Hi @nickoe ! First, thanks for your comment ! I'm running on gnuradio v 3.7.11, currently installed for using GNSS-SDR software (v0.0.11).

I have found that my input buffer sizes are not equal when I print if in the "general_work" function. It seems that there is an issue in the scheduler, but I can't figure what is it...

nickoe, 2020-04-07:
Not to be rude, you did explain your problem very well, but please show some effort into searching the issue tracker when not using the latest releases of the branches. I am pretty sure this is resolved by upgrading to 3.7.14.0. See https://github.com/gnuradio/gnuradio/issues/2530

closed 2020-04-07 by nickoe

Math32-ui, 2020-04-07:
@nickoe, thanks for your help ! I have followed the datails of this post and I can't figure how to install the 3.7.14.0, I only can install the 3.7.13.5 one...

dkozel, 2020-04-08:
@Math32-ui Are you trying to install packaged versions of GNU Radio and GNSS-SDR?

You can probably uninstall your current one and install using the GNU Radio PPA. Then you will have to build GNSS-SDR from source. Their latest release supports GNU Radio 3.8.

https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation

https://gnss-sdr.org/build-and-install/

If you hit difficulties with this the best place to get help is going to be Slack/IRC channel or discuss-gnuradio mailing list.

Math32-ui, 2020-04-09:
Hi @dkozel ! Thanks for your help. I have reinstalled a virtual machine with the lastest gnuradio release (3.8.1) and reinstall the GNSS-SDR according to the installation procedure available on its website (exluding gnuradio-dev package only).
I have the same behaviour... I can't understand why the second added source is not working has the first one in my block, whereas I duplicate all the input variables in the forecast and consume functions.
Do you know if other parameters have to be checked?
Can you think that my virtual machine is not powerful enough to deal with to sources?

Thanks again !