gnuradio/gnuradio #3323: Can't connect UDP Sink and Source on two LAN machines

andrasfuchs, 2020-04-08:
I use GNU Radio 3.7 on both a Windows 10 machine (version: 3.7.13.5, IP: 192.168.1.30) and a Raspberry Pi (version: 3.7.10 IP: 192.168.1.36).

Both have the same Flow Chart with a Signal Source -> UDP Sink, and a UDP Source -> Scope Sink setup.  They have the same sampling (1200 Hz), payload (1472), port (8843) parameters.
[image]

If I try to use the Windows machine as the source (like above), I get the following error when I run the Python script on the Raspberry Pi:

> Traceback (most recent call last):
>   File "/home/pi/Work/gr-ad2/top_block.py", line 116, in <module>
>     main()
>   File "/home/pi/Work/gr-ad2/top_block.py", line 110, in main
>     tb = top_block_cls()
>   File "/home/pi/Work/gr-ad2/top_block.py", line 65, in __init__
>     self.blocks_udp_source_0 = blocks.udp_source(gr.sizeof_float*1, '192.168.1.30', 8843, 1472, False)
>   File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 2360, in make
>     return _blocks_swig5.udp_source_make(itemsize, host, port, payload_size, eof)
> RuntimeError: bind: Cannot assign requested address

If I try the opposite, so the Raspberry Pi is the source, I get this error on Windows:
> Traceback (most recent call last):
>   File "C:\Work\gr-ad2\top_block.py", line 113, in <module>
>     main()
>   File "C:\Work\gr-ad2\top_block.py", line 107, in main
>     tb = top_block_cls()
>   File "C:\Work\gr-ad2\top_block.py", line 64, in __init__
>     self.blocks_udp_source_0 = blocks.udp_source(gr.sizeof_float*1, '192.168.1.36', 8843, 128, False)
>   File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\blocks\blocks_swig6.py", line 1252, in make
>     return _blocks_swig6.udp_source_make(itemsize, host, port, payload_size, eof)
> RuntimeError: bind: The requested address is not valid in its context

Everything work fine, if I run both the UDP Sink and Source on the same machine (using the loopback adapter).

velichkov, 2020-04-08:
Hi @andrasfuchs,

Change the IP Address in the UDP Source block to `0.0.0.0` (all interfaces) and in the UDP Sink block always specify the IP address of the other machine - for example when running the Sink on the RPi put the IP of the Windows machine and vice versa.

andrasfuchs, 2020-04-10:
Thanks a lot @velichkov, it works!

closed 2020-04-10 by andrasfuchs