gnuradio/gnuradio #3327: Error while building GNU Radio from master branch

alekhgupta1441, 2020-04-09:
I was building GNU Radio from master branch on Ubuntu 18.04 for testing my PR.
Some changes in gr-digital may be causing this. 
[image]
[image]

mormj, 2020-04-10:
This looks like a swig issue.  It happens when  the api is changed and cached swig-generated files are used as part of the compile.  Try a 'make clean' then make again.

marcusmueller, 2020-04-10:
Agreeing with @mormj 100% here. Closing unless not solved by completely emptying the build directory.

closed 2020-04-10 by marcusmueller