gnuradio/gnuradio #3362: Convert remaining printf debugging to GNU Radio logging

marcusmueller, 2020-04-13:
After #3339, we still have a few places in-tree where things are just printed out to stdout. 

Seek and destroy!

japm48, 2020-04-14:
For reference: https://gist.github.com/japm48/2f89966a03a4bcfb1f050e536340a297/raw/c804a1f5758952251a96a3572370e7b4f15ecd7a/search.txt

Although many of those lines are part of comments that are should probably be removed anyways.

BTW @marcusmueller, what are your thoughts on replacing `boost::format` with something like `fmt` ([link](https://github.com/fmtlib/fmt))? I know it would require yet another dependency and that it requires C++14 (so it should be done far in the future), but it's small enough and pretty handy.

Also is there a policy on `throw`ing vs logging + `exit(-1)`?

dkozel, 2020-04-14:
Marcus gets the final call, but we already require boost in other parts of GNU Radio that aren't likely to be easy to remove, adding another dependency and having to refactor seems like probably not worth it. We will likely have C++14 on master very soon though.

drmpeg, 2020-04-15:
It should be noted that the file lib/atsc/atsc_viterbi_gen.cc (with all the printf's) isn't really part of GNU Radio. I eliminated it's usage with this commit.

https://github.com/gnuradio/gnuradio/commit/5af80cc3146a908f3d3fd4fb201f8eebe03eff1e

It's only in the tree for legacy purposes.

adhorden, 2020-04-15:
I would like to get started on this and I feel this is a great first issue for me to start contributing to GNU Radio 🤓 .

I have had a good grep through the code and found many of the places that need refactoring so I will get started with these in a branch. I will follow the same convention as the original pull request.

85ayush, 2021-02-23:
I would love to work on this issue, I am new to open source contribution, can you please guide me?

closed 2021-03-22 by marcusmueller