gnuradio/gnuradio #3366: PMT: Serialization using endian.h does not work on windows

mrjacobagilbert, 2020-04-14:
The move to speed up serialization using endian.h in #3320 appears to be broken on windows as endian.h does not exist.

hcab14, 2020-04-14:
Maybe one of the following can help?
https://github.com/mikepb/endian.h
https://github.com/majn/tgl/blob/master/tl-parser/portable_endian.h
https://github.com/search?q=endian.h

I have no idea of how correct they are though.

mrjacobagilbert, 2020-04-14:
im not much help here with anything windows unfortunately. i can submit a PR to revert that change, or modify that change so it just fixes the f32/c32 using f64/c64.

citizenrich, 2020-04-15:
I’m also seeing this error on a build from master on Mac. Not using macports though. Do you see this on macports @michaelld ?

mrjacobagilbert, 2020-04-15:
this will be moved to boost::endian soon; you should be able to revert without too much pain.

`git revert c1c56d2d149fa1663e0d41b7f50ad60f67258ce9`

sorry about that.

mrjacobagilbert, 2020-04-15:
Solved in PR #3369 - backport updated also.

michaelld, 2020-04-23:
Yes this is an issue for OSX too: https://github.com/gnuradio/gnuradio/issues/3401

michaelld, 2020-04-23:
This is why we need -any- functional CI/QA for OSX and Windows for PRs. It doesn't have to be amazing; it just has to be sufficient.

michaelld, 2020-04-23:
Closed by #3369 .

closed 2020-04-23 by michaelld