gnuradio/gnuradio #3367: GNURadio Pip Installer

TheWylieStCoyote, 2020-04-14:
Be able to install GNURadio using pip.

dkozel, 2020-04-14:
I guess I don't know enough about pip to know if this is possible, reasonable, or a good idea? There are so many non-Python dependencies and a quick look didn't lead to any obvious examples of pip packages that depend on, for instance, boost.

I'm not against this, just confused/knowledgeable.

TheWylieStCoyote, 2020-04-14:
Marcus and I had a call a couple hours ago and came up with this idea. This seems possible as Numpy, Scipy, and some other packages have dependencies that are more complicated than GNU Radio.  I don't know of any python package that uses Boost however this is a limitation of my knowledge.

japm48, 2020-04-15:
I think this is a good idea and there are even some non-Python programs that do this, e.g. [cmake](https://pypi.org/project/cmake/). Unfortunately, I don't think it is reasonable to do on Windows.

I tried to do something similar and the problem is, of course, the dependencies. I think the worst offender is GTK (and pygobject, the GTK3 successor of PyGTK; [not available](https://pypi.org/project/PyGObject/#files) as a precompiled package), which brings a whole lot of dependencies (starting with `gettext`), that need to be heavily patched to work in Windows. The python version needs to use the same compiler (MSVC vs MinGW); note that [python.org](https://python.org/) prebuilt installers default to 32-bit and only provide MSVC version.

In Linux, it is also not ideal because users would be required to install packages from the distro's package manager. No idea of the situation in macOS.

I managed to make it work under Msys2 (using MinGW64 toolchain) and I will report back when I have time to reproduce it and write instructions. The good thing is that it uses a package manager (`pacman`, from Arch Linux) to install the toolchain so it is very easy to install the patched dependencies (see https://packages.msys2.org/), or even recompiling them, and it is reasonably well maintained. I managed to compile GR 3.9, with almost everything turned on (except ctrlport, uhd, network and sdl); note that my resulting msys2 folder size is >5 GB.

An option is then to compile on Msys2 and extract only the minimum required for GNU Radio to run (100's of MBs, probably). I believe this is the method that [Git for Windows](https://github.com/git-for-windows/git-sdk-64) uses for distribution. The only problem that I can think of is that many users expect an MSVC version and, as with everything to do with MinGW/Msys/Cygwin, this middle ground between Windows and Unix can be an endless source of headaches.

Another option I tried on Windows is using [vcpkg](https://github.com/microsoft/vcpkg) with Visual Studio, but there are still some issues (see https://github.com/microsoft/vcpkg/pull/3105). It is apparently Linux-compatible, but I believe not so well supported.

And also, consider [chocolatey](https://chocolatey.org/) for distribuition (Windows-only, unfortunately).

ryanvolz, 2021-05-13:
Just stumbled upon this while searching bugs for other things, but I think I can make a relevant comment here. Generally, I'd now consider the conda packages that I do as a suitable alternative to pip, and I think that is the cleanest way to handle the use cases that led to this request. It is conda's reason for existence (packaging Python things that need C/C++/compile dependencies), after all.

**But**, it might also be feasible to distribute GNU Radio binaries on PyPi to have them installable with pip. This would likely include bundling all of the libraries that GNU Radio needs, while deferring to pip for the Python dependencies. I'm aware of [conda-press](https://github.com/conda-incubator/conda-press) as one way of making such pip-installable wheels from a conda package, but it looks like it hasn't been updated in a while and I have no idea how well it works.

NeuroForLunch, 2021-06-24:
What is the target system here?

TheWylieStCoyote, 2021-06-25:
My target was linux machines with Python virtual environments.

NeuroForLunch, 2021-06-25:
@ryanvolz is packaging it in Conda so that should take care of this or at least make it easier to create a pip package. 

I have https://github.com/NeuroForLunch/gnuradio-companion-plus which has multiple layers on different branches if you are making a linux virtual machine or running one inside of linux. Or you can copy the code from the Dockerfiles and paste it into command line with `sudo` instead of `RUN` and install it that way.

willcode, 2023-11-28:
This is unlikely to happen, since GNU Radio is not primarily a Python program. Conda, VMs, containers, etc. fill the need.

closed 2023-11-28 by willcode