gnuradio/gnuradio #3370: Build error of "no matching constructor" with tcp_connection block

citizenrich, 2020-04-15:
Hi folks,

I'm building from master on macOS and seeing a build error on a constructor. This must be weirdness with macOS as I don't see the issue building from master on Ubuntu 18 or 19. I'm happy to close this out or for other to do so immediately as its not a supported build env.

Here's the error:
```
[ 25%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/tcp_connection.cc.o
/Users/richard/src/github.com/gnuradio/gnuradio/gr-blocks/lib/tcp_connection.cc:48:29: error: no matching constructor for
      initialization of 'std::shared_ptr<char []>'
    std::shared_ptr<char[]> txbuf(new char[len]);
                            ^     ~~~~~~~~~~~~~
```

cmake options invoked:
```
cmake \
-DCMAKE_CXX_STANDARD=11 \
-DCMAKE_C_STANDARD=11 \
-DQT_INCLUDE_DIRS=/usr/local/opt/qt/lib/QtGui.framework/Versions/Current/Headers \
-DQT_LIBRARIES=/usr/local/opt/qt/lib/QtGui.framework/Versions/Current/QtGui \
-DQWT_INCLUDE_DIRS=/usr/local/opt/qwt/lib/qwt.framework/Versions/Current/Headers \
-DQWT_LIBRARIES=/usr/local/opt/qwt/lib/qwt.framework/Versions/Current/qwt \
-DGMP_INCLUDE_DIR= \
-DGMP_LIBRARY= \
-DGMPXX_LIBRARY= \
-DMPIR_INCLUDE_DIR=/usr/local/opt/mpir/include \
-DMPIR_LIBRARY=/usr/local/opt/mpir/lib/libmpir.dylib \
-DMPIRXX_LIBRARY=/usr/local/opt/mpir/lib/libmpirxx.dylib \
-DENABLE_DOXYGEN=OFF \
-DENABLE_SPHINX=OFF \
-DENABLE_PERFORMANCE_COUNTERS=OFF \
-DPYTHON_EXECUTABLE=/usr/local/bin/python3 \
../
```

Remainder of error:
```
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3756:23: note: candidate constructor not viable: no
      known conversion from 'char *' to 'std::nullptr_t' (aka 'nullptr_t') for 1st argument
    _LIBCPP_CONSTEXPR shared_ptr(nullptr_t) _NOEXCEPT;
                      ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3770:5: note: candidate constructor not viable: no known
      conversion from 'char *' to 'const std::__1::shared_ptr<char []>' for 1st argument
    shared_ptr(const shared_ptr& __r) _NOEXCEPT;
    ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3778:5: note: candidate constructor not viable: no known
      conversion from 'char *' to 'std::__1::shared_ptr<char []>' for 1st argument
    shared_ptr(shared_ptr&& __r) _NOEXCEPT;
    ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3758:18: note: candidate template ignored: requirement
      'is_convertible<char *, char (*)[]>::value' was not satisfied [with _Yp = char]
        explicit shared_ptr(_Yp* __p,
                 ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3773:9: note: candidate template ignored: could not
      match 'shared_ptr<type-parameter-0-0>' against 'char *'
        shared_ptr(const shared_ptr<_Yp>& __r,
        ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3779:52: note: candidate template ignored: could not
      match 'shared_ptr<type-parameter-0-0>' against 'char *'
    template<class _Yp> _LIBCPP_INLINE_VISIBILITY  shared_ptr(shared_ptr<_Yp>&& __r,
                                                   ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3783:34: note: candidate template ignored: could not
      match 'weak_ptr<type-parameter-0-0>' against 'char *'
    template<class _Yp> explicit shared_ptr(const weak_ptr<_Yp>& __r,
                                 ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3788:9: note: candidate template ignored: could not
      match 'auto_ptr<type-parameter-0-0>' against 'char *'
        shared_ptr(auto_ptr<_Yp>&& __r,
        ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3798:9: note: candidate template ignored: could not
      match 'unique_ptr<type-parameter-0-0, type-parameter-0-1>' against 'char *'
        shared_ptr(unique_ptr<_Yp, _Dp>&&,
        ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3807:9: note: candidate template ignored: could not
      match 'unique_ptr<type-parameter-0-0, type-parameter-0-1>' against 'char *'
        shared_ptr(unique_ptr<_Yp, _Dp>&&,
        ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3761:9: note: candidate constructor template not viable:
      requires at least 2 arguments, but 1 was provided
        shared_ptr(_Yp* __p, _Dp __d,
        ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3764:9: note: candidate constructor template not viable:
      requires at least 3 arguments, but 1 was provided
        shared_ptr(_Yp* __p, _Dp __d, _Alloc __a,
        ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3766:26: note: candidate constructor template not
      viable: requires 2 arguments, but 1 was provided
    template <class _Dp> shared_ptr(nullptr_t __p, _Dp __d);
                         ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3767:40: note: candidate constructor template not
      viable: requires 3 arguments, but 1 was provided
    template <class _Dp, class _Alloc> shared_ptr(nullptr_t __p, _Dp __d, _Alloc __a);
                                       ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3768:51: note: candidate constructor template not
      viable: requires 2 arguments, but 1 was provided
    template<class _Yp> _LIBCPP_INLINE_VISIBILITY shared_ptr(const shared_ptr<_Yp>& __r, element_type* __p) _NOEXCEPT;
                                                  ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3754:23: note: candidate constructor not viable:
      requires 0 arguments, but 1 was provided
    _LIBCPP_CONSTEXPR shared_ptr() _NOEXCEPT;
```

citizenrich, 2020-04-16:
Closing as unsupported build environment. But, if any folks see the same issue, please add a note. Happy to test solutions with folks.

dkozel, 2020-04-16:
Nope! We're interested in macOS so this definitely can stay open. Automatic testing is still something the project needs so until that's contributed reports like yours are the best way we have of finding out about issues.

Can you quickly try writing a tiny test program with the erroring line and compiling it?
```
std::shared_ptr<char[]> txbuf(new char[25]);
```

@michaelld have you encountered this?

michaelld, 2020-04-16:
@dkozel I haven't actually built master in a while. Let me get setup & do so to see what comes of it.

michaelld, 2020-04-16:
@citizenrich what does the following return for you:
```
echo "macOS $(sw_vers -productVersion) $(sw_vers -buildVersion)"; echo "Xcode $(xcodebuild -version | awk -v ORS=' ' '{print $NF}')"
```

michaelld, 2020-04-16:
and, yes, let's leave this open! I'll work with you to resolve the issue.

citizenrich, 2020-04-16:
Hi. Here we go:

```
~/src/github.com/gnuradio/gnuradio/build [master ↑·1|✔] ⎈ docker-desktop
$ echo "macOS $(sw_vers -productVersion) $(sw_vers -buildVersion)"; echo "Xcode $(xcodebuild -version | awk -v ORS=' ' '{print $NF}')"
macOS 10.15.4 19E287
xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance
Xcode
```

So, seems I installed Xcode dev tools before Xcode (I did), and that borked finding the right compilation tools?

michaelld, 2020-04-16:
If you just updated Xcode & the CLT, then you still need to run Xcode.app to get this all working.

citizenrich, 2020-04-16:
Yep! Thanks!

michaelld, 2020-04-17:
YW!

citizenrich, 2020-04-17:
Hmm. I'm getting the same error. Here's the output of the Xcode build version:
```
$ echo "macOS $(sw_vers -productVersion) $(sw_vers -buildVersion)"; echo "Xcode $(xcodebuild -version | awk -v ORS=' ' '{print $NF}')"
macOS 10.15.4 19E287
Xcode 11.4 11E146
```

Error is the same:
```
[ 25%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/tcp_connection.cc.o
/Users/richard/src/github.com/gnuradio/gnuradio/gr-blocks/lib/tcp_connection.cc:48:29: error: no matching constructor for initialization of 'std::shared_ptr<char []>'
    std::shared_ptr<char[]> txbuf(new char[len]);
                            ^     ~~~~~~~~~~~~~
```

I tried putting the line as @dkozel suggested in a simple 'hello, world' but my cpp skills are quite unskilled.

michaelld, 2020-04-19:
@citizenrich I'll get a GR 3.8.1.0 install going this coming week & can address this issue then.

citizenrich, 2020-04-19:
Excellent, thanks, looking forward to it. When I build 3.8.1 I don't see it, so it may show up on only std pointers, as 3.8.1 looks to use boost. The error I'm seeing is on 3.9 with std pointers.

michaelld, 2020-04-19:
Interesting & thx for the info about 3.8.1 ... I'd guess this is an issue from the "boost::*_ptr" -> "std::*_ptr" change. I'll also be testing out the latest GR master, which is what you're working with. If you don't need the master, go with maint-3.8 or 3.8.1.0 ... those should be stable & very usable!

michaelld, 2021-09-25:
I've built 3.8.3.1 and master recently without issues on various OSX / MacOSX / macOS. Hence, I'm closing since I think this is fixed. If someone still has this issue, please reopen and post info about what's going on.

closed 2021-09-25 by michaelld