gnuradio/gnuradio #3371: GUI layout conflict

duggabe, 2020-04-15:
With GRC 3.8.1.0 if a flowgraph has two GUI Range blocks, where the first one has no GUI hint and the second one has `gui_hint: 0,0,3,1` then the result is that the first one is displayed but does not operate. In the example given, the `Volume` does not work, but the `Frequency` does. No error messages are generated.
OS Ubuntu 19.10
also OS Raspbian Buster 2020-02-13
GR 3.8.1.0
GUI_issue.grc: https://pastebin.com/07djpQxQ
GUI_issue.py: https://pastebin.com/CSwmpftB

duggabe, 2020-04-26:
Another example of the problem is in https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/examples/qtgui_tags_viewing.grc
The Delay range GUI can be seen behind the Time Sink.

ChristopheSeguinot, 2021-02-18:
I investigated this issue. Finally found it's not related to  #4249
The problem arises from using some widget without GUI Hint, and some with GUI Hint. Widgets are inserted in a QGridLayout. Examining flowgaph generated .py file we for example get

`self.top_grid_layout = Qt.QGridLayout()
self.top_layout.addLayout(self.top_grid_layout)
...  
self.top_grid_layout.addWidget(self._volume_win)
...   
self.top_grid_layout.addWidget(self._freq_win, 0, 0, 1, 1)
`

In case some widget without GUI Hint is added first, it use (row,col) =0,0 . 
In the present code it is superposed to the widget using 0,0

I found that this can be caused by any widget, not only QTGUI Range. 
Possible solutions: 

1. all widgets must have a GUI Hint , OR all must have no GUI Hint : rejected, this is not convenient
2. all widgets without GUI Hint should be added after those having a GUI Hint (tested by modifying generated python : it works

Looking into grc/core/utils/expr_utils.py and /usr/lib/python3/dist-packages/gnuradio/gr/top_block.py I found that blocks are ordered by name, and by GUI HInt. 
We could modify this so that widgets without GUI Hint are processed after all other. 

However, it seem's that this odering is not applied before generating the flowgraph.

Questions:
is it possible to apply ordering before generating the flowgraph ?
is it possible to modifiy ordering method _get_block_sort_text whithout breaking GNURadio ? (or do we need to define  new sorting method ? )

Could someone give me some advice to fix this issue?

duggabe, 2021-03-11:
Is this ready to close now?

ChristopheSeguinot, 2021-03-11:
Yes it can be closed

closed 2021-03-16 by duggabe