gnuradio/gnuradio #3376: gr-fec  fecapi_async_decoders example not working

pepassaco, 2020-04-17:
Hello!

I was trying the example fecapi_async_decoders from gnuradio/gr-fec/examples (my GNURadio version is 3.8), surprisingly, it does not work as it should. If you try the dummy or repetition encoders, the output bytes match the input ones. However, if you select the convolutional encoder, they do not match at all:

https://imgur.com/R41BUj3

What can be causing this bug and how can it be solved? The fecapi_decoders example seems to work ok interestingly enough.

mohmmadd, 2020-04-25:
It's work in GNURadio 3.7. 
packet_loopback_hier.grc from gr-digital/examples/packet that use async-decoder/encoder has the same problem.

mormj, 2020-04-26:
Looks to be related to #2642 - pretty much looks like conv encoder examples don't work in 3.8.  This should definitely be prioritized.

mormj, 2020-04-30:
@jdemel @michaelld - I traced this back to the volk_8u_x4_conv_k7_r2_8u kernel, which behaves differently between whatever is included in 3.7 and whatever got included with 3.8.  When I manually paste the generic kernel in the cc_decoder_impl.cc code, the bits line up as expected.

jdemel, 2020-05-01:
@mormj So the generic VOLK kernel works as expected? But some other kernel version does not?
Do you see correct results with`VOLK_GENERIC=1`? This should enforce generic kernel use.

mormj, 2020-05-04:
This works - but leaving off the VOLK_GENERIC does not
```
josh@josh-clean1910:/share/gnuradio/grdev/src/gnuradio$ export VOLK_GENERIC=1
josh@josh-clean1910:/share/gnuradio/grdev/src/gnuradio$ /usr/bin/python3 -u /share/gnuradio/grdev/src/gnuradio/gr-fec/examples/fecapi_cc_decoders.py
```
This is the latest master and latest volk.  Looks like an AVX implementation was added in volk between the 3.7 and 3.8 release

Here are my machines in cmake
```
-- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
-- Available machines: generic;sse2_64_mmx;sse3_64_mmx;ssse3_64_mmx;sse4_a_64_mmx;sse4_1_64_mmx;sse4_2_64_mmx;avx_64_mmx;avx2_64_mmx;avx512f_64_mmx;avx512cd_64_mmx
```

michaelld, 2020-05-04:
@mormj which version of VOLK are you using for testing? If not the current release, can you try the current release?

mormj, 2020-05-05:
This was with the latest tip on master:

```
commit 6acb788a50087b74783670e5c1cd17fddd681953 (HEAD -> master, origin/master, origin/HEAD)
Merge: 4a196aa 930aa21
Author: Michael Dickens <michael.dickens@ettus.com>
Date:   Sat May 2 20:46:52 2020 -0400

    Merge pull request #378 from jdemel/ci-fix-remove
    
    ci: Remove Ubuntu 16.04 GCC5 test on TravisCI
```

But I know for sure it's broken before that.  I'll try the last tagged release

mormj, 2020-05-05:
The issue still occurs with Volk 2.2.1, and goes away with VOLK_GENERIC=1

michaelld, 2020-05-18:
> The issue still occurs with Volk 2.2.1, and goes away with VOLK_GENERIC=1

How about with the recent Volk 2.3.0 release?

mohmmadd, 2020-05-19:
I checked it. The problem persists.

mohmmadd, 2020-07-30:
Is problem resolvable?!

mormj, 2020-07-30:
I'm sure it is, but it will require a fix in Volk, or removal of the specific kernel

jdemel, 2020-07-30:
This is a really tricky one. I ran `volk_profile` and now I can't reproduce this error.

tim-huggins, 2020-08-06:
Is there a workaround for this aside from reverting to 3.7?

jdemel, 2020-08-07:
@tim-huggins, yeah. there's a workaround. Run `volk_profile`, then inspect `~/.volk/volk_config`. Find the line that starts with:
```
volk_8u_x4_conv_k7_r2_8u 
```
and make sure,  it ends with `spiral spiral` (or `generic generic` or a mix thereof)
```
volk_8u_x4_conv_k7_r2_8u spiral spiral
```
The AVX kernel seems to be the reason for this bug. Unfortunately, it only manifests itself in GR. Thus, this bug is difficult to reproduce and track down.

Aang23, 2022-02-07:
With the AVX kernel disabled in Volk, and https://github.com/gnuradio/volk/pull/475, this issue appears to be gone.

closed 2022-02-07 by michaelld