gnuradio/gnuradio #3398: GRC: handling of certain values within variables

mrjacobagilbert, 2020-04-21:
From a blank flowgraph, changing a variable to `pmt.init_u8vector(1,[203])` will cause the following:

```
UnicodeEncodeError: 'utf-8' codec can't encode character '\udccb' in position 30: surrogates not allowed
Traceback (most recent call last):
  File "/opt/gnuradio/lib/python3/dist-packages/gnuradio/grc/gui/ParamWidgets.py", line 124, in _apply_change
    self._update_gui()
  File "/opt/gnuradio/lib/python3/dist-packages/gnuradio/grc/gui/ParamWidgets.py", line 94, in _update_gui
    self.set_tooltip_text(self.param.format_tooltip_text())
  File "/opt/gnuradio/lib/python3/dist-packages/gnuradio/grc/gui/ParamWidgets.py", line 156, in set_tooltip_text
    self._input.set_tooltip_text(text)
UnicodeEncodeError: 'utf-8' codec can't encode character '\udccb' in position 30: surrogates not allowed
```

Also, PMT is not a default import, and it probably should be...

mrjacobagilbert, 2020-04-21:
This is on both master and maint-3.8.

ThomasHabets, 2020-11-16:
This seems to me like a bug in Python code. With Python3 it should use type `bytes` instead of `str`, right?

PR #3916 (2020-12-19): pmt: use to_string for char string_refs
The current method of converting pmt char values to strings results in non-unicode characters - looks like it was using the pointer value in the string reference.  This just uses std::to_string on the char values

Fixes #3398
closed 2020-12-19 by marcusmueller