gnuradio/gnuradio #3405: pybind11: Check for correct version.

drmpeg, 2020-04-24:
pybind11 2.0.1 (the version packaged with Ubuntu 18.04) is missing pybind11/embed.h. Compilation fails at:
```
/home/ubuntu/xfer/gnuradio/gnuradio-runtime/lib/block_gateway_impl.cc:11:10: fatal error: pybind11/embed.h: No such file or directory
 #include <pybind11/embed.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

mormj, 2020-04-24:
Thanks!  Most distros have pybind11 2.4.3 and I've specified that as the required minimum.  So I guess the action here is to clean up the CMake to actually check the version.  We can put the newer pybind in the ppa

skoslowski, 2020-04-24:
Why don't we just vendorize pybind?

mormj, 2020-04-24:
It's certainly possible since its all header and has no dependencies - it's how @mbr0wn handled it for pyuhd.  We were trying to avoid it since it would be a departure from how other GR dependencies are handled, and it is packaged for most distros we're aware of.  But it's a very fair question.

noc0lour, 2020-04-24:
Would be neat to check if we could drop it in tree and use it for situations where pybind version is smaller than 2.4.3, as it's header only and if we do it right there is almost no harm :) Installing headers so OOTs have access to Pybind we need to handle with caution.

mormj, 2020-04-24:
The GR binary installers could be tricky in that case.  Could we install to /usr/include/gnuradio/pybind11 and change the include path in the cmake for OOTs if the version is less than the minimum?

noc0lour, 2020-04-24:
In the gnuradioConfig.cmake you can flip the switch I would think. And as every OOT will have to `find_package(Gnuradio)` we can point to whatever pybind we want.

skoslowski, 2020-04-24:
While one could do that and write some clever cmake, it would still be more complicated than using a single fixed version... Since we don't really need pybind binding for external libs, there aren't any limitations

noc0lour, 2020-04-24:
The only real limitation is to provide it for downstream OOTs, so making sure they use the PyBind11 version we ship. But that's about it.

skoslowski, 2020-04-24:
true, exporting these headers should not be an issue.

closed 2021-01-19 by drmpeg