gnuradio/gnuradio #3408: pybind11: compile fails on 32-bit system.

drmpeg, 2020-04-24:
I'm going to guess the use of "long" is problematic.
```
/home/ubuntu/xfer/gnuradio/gnuradio-runtime/python/pmt/bindings/pmt_python.cc: In function ‘void bind_pmt(pybind11::module&)’:
/home/ubuntu/xfer/gnuradio/gnuradio-runtime/python/pmt/bindings/pmt_python.cc:773:90: error: address of overloaded function with no contextual type information
    size_t, std::vector<unsigned long, std::allocator<unsigned long>> const&)) &
                                                                               ^

    ::pmt::init_u64vector,
    ~~                                                                          

/home/ubuntu/xfer/gnuradio/gnuradio-runtime/python/pmt/bindings/pmt_python.cc:788:83: error: address of overloaded function with no contextual type information
       (pmt::pmt_t(*)(size_t, std::vector<long, std::allocator<long>> const&)) &
                                                                               ^

           ::pmt::init_s64vector,
           ~~                                                                   

/home/ubuntu/xfer/gnuradio/gnuradio-runtime/python/pmt/bindings/pmt_python.cc:1178:28: error: address of overloaded function with no contextual type information
               pmt::pmt_t)) &
                            ^
               ::pmt::u64vector_elements,
               ~~            
/home/ubuntu/xfer/gnuradio/gnuradio-runtime/python/pmt/bindings/pmt_python.cc:1185:75: error: address of overloaded function with no contextual type information
           (std::vector<long, std::allocator<long>> const (*)(pmt::pmt_t)) &
                                                                           ^
               ::pmt::s64vector_elements,
               ~~                                                           
[ 14%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/sptr_magic.cc.o
gnuradio-runtime/python/pmt/bindings/CMakeFiles/pmt_python.dir/build.make:65: recipe for target 'gnuradio-runtime/python/pmt/bindings/CMakeFiles/pmt_python.dir/pmt_python.cc.o' failed
make[2]: *** [gnuradio-runtime/python/pmt/bindings/CMakeFiles/pmt_python.dir/pmt_python.cc.o] Error 1
CMakeFiles/Makefile2:4813: recipe for target 'gnuradio-runtime/python/pmt/bindings/CMakeFiles/pmt_python.dir/all' failed
make[1]: *** [gnuradio-runtime/python/pmt/bindings/CMakeFiles/pmt_python.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

drmpeg, 2020-04-24:
This is with pybind11 version 2.4.3.

mormj, 2020-04-26:
Thanks for catching this one.
Should be fixed with https://github.com/gnuradio/gnuradio/commit/473ee67e2cbf95da71eec5b7a3649807b1dbec7c

commit 473ee67e (2020-04-26): pmt: replace "long" with int64_t in python bindings

pygccxml seems to replace int64_t and uint64_t with long and unsigned
long.  This causes a mismatch in the overload on 32b systems

Fixes #3408
closed 2020-04-26 by drmpeg