gnuradio/gnuradio #3411: Error setting frequency with a message

duggabe, 2020-04-26:
While testing https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/examples/test_qtgui_msg.grc I found that changing frequency either by entering a value in the Message Edit Box or by double-clicking a frequency on either Sink trace caused the frequency to change to twice the value entered. For example: entering '2500' in the Message Edit Box displays a signal at 5000 Hz. It does the same for double-clicking a value on the x-axis.

I moved the input of the Message Debug block to obtain the message sent. It shows:
<pre>
******* MESSAGE DEBUG PRINT ********
(freq . 2500)
************************************
</pre>
Values with decimal points behaved the same way.
Tested with Raspian Buster with 3.8.1.0; and Ubuntu 20.04 with 3.9.0.0-git

mrjacobagilbert, 2020-04-27:
This is because that particular flowgraph is sending the set_freq message to both the signal source and the qt sinks which treat this information differently. The signal source now generates a 2.5kHz tone instead of 1kHz _and_ the qt sinks are being told to use 2.5 kHz as the new center frequency. 

One or the other should be used, not both. The blocks are all doing the correct thing its just a mistake in the flowgraph (which should be fixed).

duggabe, 2020-04-27:
@mrjacobagilbert I confirm that removing those links fixes the problem.

I am also working on https://github.com/gnuradio/gnuradio/issues/1971 and will be generating a PR. Should I leave this issue open and put a note that the PR will close this?

mrjacobagilbert, 2020-04-27:
I would leave it open for now. I believe the intention was to demonstrate the 'Simple Spectrum Analyzer' functionality where you use the QT GUI to tune a USRP. It's just that the signal source doesnt quite provide the exact functionality. Give this a shot:

[image]

closed 2020-06-11 by 777arc