gnuradio/gnuradio #3413: macOS: Delete key doesn't delete block

samyk, 2020-04-26:
gnuradio 3.7.13.5 on XQuartz 2.7.11 and macOS 10.14.6 installed from MacPorts.

"Delete" key doesn't delete a selected block, though that is what is noted as the key to delete if right clicking a block. fn+delete keys combined do delete, however, which is not typical behavior.

NeuroForLunch, 2021-06-24:
More info: This affects all Macs and all versions of GNURadio. The delete key on a macbook  is being read as `delete_or_backspace` and doesn't work but fn+delete generates `delete_forward` and deletes a block.

luzpaz, 2023-07-22:
Needs `macos` label

marcusmueller, 2023-07-22:
Since this is the 3.7 series, and we've currently (and foreseeable) don't have the resources to maintain, fix and update that, it's considered End-of-Life. And so I'm afraid I'll close this issue.

closed 2023-07-22 by marcusmueller