gnuradio/gnuradio #3414: macOS: Shift+tab does not go to previous input

samyk, 2020-04-26:
gnuradio 3.7.13.5 on XQuartz 2.7.11 and macOS 10.14.6 installed from MacPorts.

If I double click a block in GRC, hitting Tab properly goes through the input boxes. However, shift+tab natively goes in reverse but does not work here.

Additionally, I can use the "down" arrow to go through inputs but this is a problem once it hits a dropdown, at which point the down arrow changes the selected value, and once it's hit the last item in the dropdown, "down" no longer goes down and you're forced to use tab or click. I'd suggest not having "down" switch between inputs as it's odd behavior, but it should continue to allow selecting different items in a dropdown which is normal behavior on most OSs.

Thanks!

luzpaz, 2023-07-22:
needs `macos` label

marcusmueller, 2023-07-22:
Same as your other 3.7 issue: I'm terribly sorry!

closed 2023-07-22 by marcusmueller