gnuradio/gnuradio #3415: macOS: Audio Source doesn't work until Microphone permission is given

samyk, 2020-04-26:
gnuradio 3.7.13.5 on XQuartz 2.7.11 and macOS 10.14.6 installed from MacPorts.

Audio Source appears to work and sends samples, but the samples contain no audio.

This is resolved once I go to System Preferences -> Security & Privacy -> Privacy -> Microphone and added "iTerm2" (the terminal app I'm running GRC from), then restart XQuartz / GRC. It's unclear as there is no error.

I tried updating the wiki by requesting an account and confirming my email, but the site is not sending me a password, nor is the "Forget password" working for me - wanting to update
https://wiki.gnuradio.org/index.php/Audio_Source

Additionally that wiki has incorrect information for Input -- if I actually adjust the inpu device name to "Internal Microphone" (as noted in my System Preferences->Sound->Input area), the block fails entirely as it's expecting "Built-in Microphone" which is not the same name as in preferences.

Also when the wiki sent me an email to confirm my account, it includes "Someone, probably you from IP address 172.18.0.7" but that's not public IP address space -- maybe the wiki is not reading X-Forwarded-For header or maybe real IP isn't being passed along.

Thanks!

warnes, 2020-05-28:
Thanks for posting this.  I was also puzzled about why the Audio Source was generating only zeros.

NeuroForLunch, 2021-06-24:
# Solution 1
Open automator [image]

Create an app called GNURadio and just have it run the shell script gnuradio-companion

[image]

Get the notification
[image]

# Solution 2
`brew install gnuradio` will install GNURadio 3.9.2 and audio should work no problem or you can follow the steps from Solution 1 and it will definitely ask for the permission.

luzpaz, 2023-07-22:
needs `macos` label

marcusmueller, 2024-01-22:
I don't have any Mac OS machine to reproduce this – help from someone with one utterly needed.

samyk, 2024-01-24:
I *disabled* iTerm from my `Privacy & Security->Microphone`, restarted iTerm, then ran `/opt/homebrew/bin/gnuradio-companion` (3.10.7.0 w/python 3.11.7 on macOS 14.1.1) and get this:

```
Loading: "/Users/samy/Documents/Code/gnuradio/repro3451.grc"
>>> Done

Generating: '/Users/samy/Documents/Code/gnuradio/repro3451.py'

Executing: /opt/homebrew/Cellar/gnuradio/3.10.7.0_4/libexec/venv/bin/python -u /Users/samy/Documents/Code/gnuradio/repro3451.py

log :warning: Matplotlib is a required dependency to use DistanceRadar and AzElPlot.  Please install matplotlib to use these blocks (https://matplotlib.org/)
audio_osx_source :info: 

Using input audio device 'MacBook Pro Microphone'.
  ... which is the current default input audio device.
  Changing the default input audio device in the System Preferences will 
  result in changing it here, too (with an internal reconfiguration).

Failed to set window logo
```
[image]

If I then add the microphone permission back to iTerm, restart iTerm, and restart GRC, it works as expected:
```

Loading: "/Users/samy/Documents/Code/gnuradio/repro3451.grc"
>>> Done

Generating: '/Users/samy/Documents/Code/gnuradio/repro3451.py'

Executing: /opt/homebrew/Cellar/gnuradio/3.10.7.0_4/libexec/venv/bin/python -u /Users/samy/Documents/Code/gnuradio/repro3451.py

log :warning: Matplotlib is a required dependency to use DistanceRadar and AzElPlot.  Please install matplotlib to use these blocks (https://matplotlib.org/)
audio_osx_source :info: 

Using input audio device 'MacBook Pro Microphone'.
  ... which is the current default input audio device.
  Changing the default input audio device in the System Preferences will 
  result in changing it here, too (with an internal reconfiguration).
```
[image]