gnuradio/gnuradio #3419: Probe Signal block does not display ID

SlavDimov, 2020-04-28:
The probe signal block in version 3.8.1.0 (from copr repo)
does not display an ID for the Probe Signal block. The only way to access it is by opening the .grc yaml file.
[image]

mormj, 2020-04-29:
There is a grc option that you can check - "show all block ids" - that should enable you to see the ID.  But this still is a bug that needs to be fixed.  I believe it has been fixed and perhaps just not backported to 3.8

SlavDimov, 2020-04-29:
I didn't know that option. Thanks!

antonjan, 2021-05-20:
Same problem and dont seem to find the "Show all block Id's"

antonjan, 2021-05-20:
A quick solution is to open blocks_probe_signal_x.block.yml and blocks_probe_signal_vx.block.yml and add show_id to the flags.

flags: [ show_id, python, cpp ]

dkozel, 2021-05-24:
@willcode Possibly the commit to always show block ids hasn't been backported to 3.8?

willcode, 2021-05-24:
Backported in 3.8.3.0 ... this change was way after the OP's 3.8.3.1.

Ref: https://github.com/gnuradio/gnuradio/pull/4497

dkozel, 2021-05-24:
Thanks!

@antonjan The fix is available in newer releases of GNU Radio 3.8. It sounds like you've solved the issue for yourself but if you have any issues with it in the future I'd recommend upgrading. If that doesn't solve it then please do come back to this issue and let us know. I'm closing it for now. All the best on your project.

closed 2021-05-24 by dkozel