gnuradio/gnuradio #3424: pybind: Exceptions not being handled via catch_exceptions flag

mormj, 2020-04-28:
Causing failure in qa_uncaught_exceptions.py test

This particular test case uses a python flowgraph, which is inherently different than a c++ based on.  The cause lies somewhere in the way the new block_gateway_impl.cc handles (or doesn't handle?) the exception that is called in the general_work in the python code.

jwmelto, 2022-10-04:
Looking at the python, I don't understand why the flowgraph instantiates two different `execpt_block` instances, This seems to guarantee that one of them will trigger the opposite case being tested for.

Or, the issue might be that `except_block` ignores the `throw_except` argument. Maybe it should only throw its exception if the parameter is `True`?

marcusmueller, 2024-01-28:
Test case is fixed in 7eced84a812fac4b4ed2aea421694f63e9f3c91b

closed 2024-01-28 by marcusmueller