gnuradio/gnuradio #3428: No package 'mpir 3.0', 'thrift', 'jack', 'portaudio-2.0',  found. Could NOT find LIBCODEC2. gnuradio-companion disabled.

sahilc556, 2020-04-29:
- [ ] root@kali:/usr/local/src/gnuradio/build# cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 ../
- [ ] -- Build type set to Release.
- [ ] -- 
- [ ] -- The build system will automatically enable all components.
- [ ] -- Use -DENABLE_DEFAULT=OFF to disable components by default.
- [ ] -- 
- [ ] -- Configuring testing-support support...
- [ ] --   Enabling testing-support support.
- [ ] --   Override with -DENABLE_TESTING=ON/OFF
- [ ] -- Extracting version information from git describe...
- [ ] -- Compiler Version: cc (Debian 9.3.0-10) 9.3.0
- [ ] Copyright (C) 2019 Free Software Foundation, Inc.
- [ ] This is free software; see the source for copying conditions.  There is NO
- [ ] warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
- [ ] -- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
- [ ] /usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
- [ ] -- ADDING PERF COUNTERS
- [ ] -- User set python executable /usr/bin/python3
- [ ] -- Python checking for six - python 2 and 3 compatibility library - found
- [ ] -- 
- [ ] -- Checking for module SWIG
- [ ] -- Found SWIG version 4.0.1.
- [ ] -- 
- [ ] -- Configuring python-support support...
- [ ] --   Dependency PYTHONLIBS_FOUND = TRUE
- [ ] --   Dependency SWIG_FOUND = TRUE
- [ ] --   Dependency SWIG_VERSION_CHECK = TRUE
- [ ] --   Dependency SIX_FOUND = TRUE
- [ ] --   Enabling python-support support.
- [ ] --   Override with -DENABLE_PYTHON=ON/OFF
- [ ] -- 
- [ ] -- Configuring VOLK support...
- [ ] -- Build type set to Release.
- [ ] -- Extracting version information from git describe...
- [ ] -- 
- [ ] -- Python checking for python >= 3.4
- [ ] -- Python checking for python >= 3.4 - found
- [ ] -- 
- [ ] -- Python checking for mako >= 0.4.2
- [ ] -- Python checking for mako >= 0.4.2 - found
- [ ] -- QA Testing is enabled.
- [ ] --   Modify using: -DENABLE_TESTING=ON/OFF
- [ ] -- System profiling is disabled.
- [ ] --   Modify using: -DENABLE_PROFILING=ON/OFF
- [ ] -- Compiler name: GNU
- [ ] -- Use `posix_memalign` for aligned malloc!
- [ ] -- x86* CPU detected
- [ ] -- Compiler doesn't support NEON, Overruled arch neon
- [ ] -- Compiler doesn't support NEON, Overruled arch neonv7
- [ ] -- Compiler doesn't support NEON, Overruled arch neonv8
- [ ] -- CPU width is 64 bits, Overruled arch 32
- [ ] -- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
- [ ] -- Available machines: generic_orc;sse2_64_mmx_orc;sse3_64_mmx_orc;ssse3_64_mmx_orc;sse4_a_64_mmx_orc;sse4_1_64_mmx_orc;sse4_2_64_mmx_orc;avx_64_mmx_orc;avx2_64_mmx_orc;avx512f_64_mmx_orc;avx512cd_64_mmx_orc
- [ ] -- BUILD TYPE = RELEASE
- [ ] -- Base cflags = -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall
- [ ] -- BUILD INFO ::: generic_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall 
- [ ] -- BUILD INFO ::: sse2_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2
- [ ] -- BUILD INFO ::: sse3_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3
- [ ] -- BUILD INFO ::: ssse3_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3
- [ ] -- BUILD INFO ::: sse4_a_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -msse4a -mpopcnt
- [ ] -- BUILD INFO ::: sse4_1_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1
- [ ] -- BUILD INFO ::: sse4_2_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt
- [ ] -- BUILD INFO ::: avx_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx
- [ ] -- BUILD INFO ::: avx2_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2
- [ ] -- BUILD INFO ::: avx512f_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2 -mavx512f
- [ ] -- BUILD INFO ::: avx512cd_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2 -mavx512f -mavx512cd
- [ ] -- Compiler Version: cc (Debian 9.3.0-10) 9.3.0
- [ ] Copyright (C) 2019 Free Software Foundation, Inc.
- [ ] This is free software; see the source for copying conditions.  There is NO
- [ ] warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
[... 266 lines cut ...]
- [ ] --   * * alsa
- [ ] --   * * oss
- [ ] --   * gr-channels
- [ ] --   * gr-qtgui
- [ ] --   * gr-trellis
- [ ] --   * gr-uhd
- [ ] --   * gr-utils
- [ ] --   * gr_modtool
- [ ] --   * gr-video-sdl
- [ ] --   * gr-vocoder
- [ ] --   * * gsm
- [ ] --   * gr-wavelet
- [ ] --   * gr-zeromq
- [ ] -- 
- [ ] -- ######################################################
- [ ] -- # Gnuradio disabled components                        
- [ ] -- ######################################################
- [ ] --   * gnuradio-companion
- [ ] -- 
- [ ] -- Using install prefix: /usr/local
- [ ] -- Building for version: 3.8.1.0 / 3.8.1
- [ ] -- Configuring done
- [ ] -- Generating done
- [ ] -- Build files have been written to: /usr/local/src/gnuradio/build
- [ ]

mormj, 2020-04-29:
Looks like the only thing missing to build gnuradio-companion is mako.  https://www.makotemplates.org/

nickoe, 2021-01-03:
No reply from OP in a long time. Closing.

closed 2021-01-03 by nickoe