gnuradio/gnuradio #3434: Pybind11 firdes win_type not compatible with integer index

ejk43, 2020-05-01:
Hi @mormj just tried a dummy flowgraph on pybind11 branch

[image]

QT Gui Sink indicates an error:

```
Param - Window Type(wintype):
	Expression win_type.WIN_BLACKMAN_hARRIS is invalid for type 'int'.
```

Replacing the "Window Type" box with a 0, 1, etc, makes things happy.

Block YML indicates the param wintype should be an integer... I'm not familiar with pybind at all but guessing the return type of the swigged firdes.win_type enum used to be an integer, but is now not...

```
In [4]: from gnuradio.filter import firdes                                                                                                                                                                 

In [5]: firdes.WIN_BLACKMAN_HARRIS                                                                                                                                                                         
Out[5]: win_type.WIN_BLACKMAN_hARRIS
```

mormj, 2020-05-01:
Thanks for testing this out - for some reason with the pybind bindings the enum parameters do not directly accept ints directly anymore.  So I just pushed the fixed yml file to https://github.com/gnuradio/gnuradio/commit/626770ed4def1054871c00b168a9e1038da535f0

ejk43, 2020-05-01:
I made the same change locally too, though there's one implication worth considering: using the "int" type, grc users are able to pass a local variable in to the selection box. But as far as I can tell, an enum type only allows graphical selection of the enum (can't use a programmatic selection). It's probably not an issue for this particular block but might show up in others.

mormj, 2020-05-01:
That's a good point - there's probably a way in pybind11 to configure the binding to map the int to enum so it will accept int as a parameter, but I haven't figured it out yet

mormj, 2020-05-04:
The change has been checked into pybind branch.  Will open another issue with the int --> enum mapping question

closed 2020-05-04 by mormj

dylan-baros, 2020-12-18:
I am having this same issue on 3.9.0.0. Verified that the yaml file is correct.

f6bvp, 2021-01-23:
Hi,
I have the same kind of issue with Band Pass Filter trying to execute 
Gnu Radio tutorial example : Simulation example: AM transmitter and receiver
https://wiki.gnuradio.org/index.php/Simulation_example:_AM_transmitter_and_receiver
AM_transmit blocks are all ok. AM_receive has an error with Hamming. 
Expression win_type.WIN_HAMMING is invalid for type 'int'.
I cannot see where to perform change as stated in above discussion.
Can someone explain for a newbye ?
Bernard
f6bvp

bobbryce, 2021-01-23:
Hi there,
Similar issues here.
Brand new Linux Mint 20.1, plus fresh installation using PPA master (sudo add-apt-repository ppa:gnuradio/gnuradio-master).

Simple flowdiagram that worked in 3.8.2 shows unexpected errors in v3.9.0.

[image]

1. Signal Source complains: Expression gr_waveform_t.GR_SIN_WAVE is invalid for type 'int'.
2. QT GUI Waterfall Sink complains: Param - Window Type(wintype):
	Expression win_type.WIN_BLACKMAN_hARRIS is invalid for type 'int'.

Any thoughts?

duggabe, 2021-01-23:
It's a nuisance, but the fix is not too hard:
Any time a filter, waterfall, or other block has a filter type parameter, open the Properties for that block, click on the window type and reselect the filter (such as Hamming or  Blackman-harris) then Apply and OK.
There was an enumerated type change between 3.8 and 3.9 which caused this.
The same problem happens going the other direction: 3.9 to 3.8. Also same fix.

bobbryce, 2021-01-23:
duggabe,
Appreciate you coming back with a suggestion. However, this does not work in my situation - 
a) I may not have understood your definition of a filter block, but neither of the two blocks flagged in 'red' in the diagram has any obvious 'filter properties' - in particular the Signal Source? - and 
b) yes - the waterfall block has a window type setting, but changing this as suggested - either back to itself or for another window type - makes no difference to the error flagged above.

I have attached the output from the installation, just in case it flags anything obviously wrong. There are no explicit errors that I could see.

[Build_history_masterPPA_210123.txt](https://github.com/gnuradio/gnuradio/files/5861194/Build_history_masterPPA_210123.txt)

f6bvp, 2021-01-23:
Hi,

I confirm what is reported here. Reseting Hamming window does not change 
anything.

Bernard F6BVP / AI7BG

http://radiotelescope-lavillette.fr

duggabe, 2021-01-24:
@bobbryce and @f6bvp If you are comfortable with posting your flowgraphs to Pastebin or similar site, I can look at them to determine the problems. Just give the link here.

bobbryce, 2021-01-24:
duggabe
This is a link to the above flow diagram. Hope that you can access this as never tried this service before.....
http://ge.tt/6wtslZA3

Bob

f6bvp, 2021-01-24:
Hi,

@bobbryce already sent a simple flowgraph.

Here is a link for AM_receive.grc I wrote based on former GNU tutorial 
AM receive and transmit.

The FIR filter displays the same issue and error message as reported by 
@bobbryce.

I also provide the link for AM_transmit.grc.

http://f6bvp.org/AM_receive.grc

http://f6bvp.org/AM_transmit.grc

Bernard
F6BVP / AI7BG

http://radiotelescope-lavillette.fr

duggabe, 2021-01-24:
Bernard:
Change Band Pass Filter to 'Hamming'  Click on the block; click on 'Properties'; for the 'Window', pull down and choose 'Hamming' then Apply and OK.
Change Multiply Const to 'volume', using the 'volume' variable from the GUI Range. That is what ties the GUI to the Multiply.

Bob:
The same problem as I described above applies to other blocks with enumerated types. :(  So change Signal Source to 'Sine' and change Waterfall to 'Blackman-harris'.
BTW, I couldn't download your file. You may not have marked it Public. I just rebuilt your flowgraph from the image above.

bobbryce, 2021-01-24:
Hi duggabe.
Again my thanks for helping - but no joy. I've tried these suggestions already.
Changing Cosine to Sine in Signal Source and clicking Apply/Close or just Close, leaves the Block tagged RED.
The Waterfall is already set to Blackman-Harris, but re-setting to this, or setting to any of the other options and trying Apply/Close or just Close leaves the Block tagged RED and the error message unchanged.

Deleting (for example) the Signal Source Block and selecting a new one brings in a Block that is already tagged RED (I think this would be normal) - but connecting this to the Throttle does not remove the RED or the Error.

I tried again with the download - there was a suggestion online that the annonymous operation no longer works and so I took out an account - as a side issue I would be interested to know if this works or not. 
http://ge.tt/6wtslZA3

Thanks again
Bob

duggabe, 2021-01-24:
@bobbryce I still could not download your file. I have not tried 'ge.tt' before. Here is the file I built for you using GR 3.9.0.0 built from maint-3.9
  https://pastebin.com/BqCax4CG

bobbryce, 2021-01-24:
duggabe,
Thanks for taking the time and effort to rebuild my script.
Guess what - exactly the same symptoms on my machine as before - see snapshot below.
Assuming this runs fine on your system, it points to something strange in either the PPA master, or associated with Linux Mint 20.1, or some weird conflict between the two.

[image]

Can you think of any diagnostics I can run or any deeper system exploring that might point the finger of suspicion somewhere?

I am getting somewhat dispirited now, having bought a Pluto for Christmas and not yet getting a build that works both with GRC and with Python. So far, I have been assuming the fault lay with a conflict in iio between Analog Devices and GNU Radio, but the issues experienced go far wider than that.

I have so far tried loading GRC v3.7, 3.8.1, 3.8.2 and 3.9.0 and have also tried different build methods including use of 'apt-get', PPA, and build from sources. Every single build has resulted in problems of some kind. I did manage to get Pluto scripts running in v3.8.2 and in v3.9.0 built from source, but could never get Python access to work at the same time.

PS Thanks for feedback on Ge.tt - I guess I will ditch that and try pastebin!

Bob

duggabe, 2021-01-25:
Let's take this offline.
Look at https://wiki.gnuradio.org/index.php/Talk:HamRadio#Questions and send me an email. I can send you some things to try.
- Barry E Duggan (duggabe)

Jim2435074, 2021-02-12:
duggabe -

If you figure this out, Please POST A SOLUTION! I'm on Ubuntu 20.04, clean install on Mac (parallels, Mojave, python 3.8.5) with a goal of getting Adalm Pluto working ... no joy so far. Same problems with window type and signal type... tried gnuradio PPA installs for 3.9, 3.9-dev, 3.8 an 3.7 with a fresh vm for all attempts. (multiple problems trying to get gr-iio working as well) Any help would be appreciated! 

cheers,
-jim

duggabe, 2021-02-12:
@Jim2435074 : I'm confused by your statement "I'm on Ubuntu 20.04, clean install on Mac".
Here is a procedure which should work, including getting gr-iio going: https://wiki.gnuradio.org/index.php/User:Duggabe

Jim2435074, 2021-02-12:
duggabe -
sri for the confusion. Im using a MacBookPro w/Mojave and a VM Ubuntu 20.04. Anyway, so I started a fresh VM and was careful to watch the /usr and /usr/local paths in PYTHONPATH and LD_LIBRARY_PATH. That seems to have fixed the problems with enum/int in window type and signal type. The build directions for gr-iio were very clear and really helped... gr-iio installed and seems to work! Thanks for the help!
cheers
-jim

Abraxas3d, 2021-05-21:
Same problem today. 

Ubuntu 20.04
GNU Radio 3.10