gnuradio/gnuradio #3435: Using 2 Python Blocks Cause the Flowgraph to Crash

giladbeeri, 2020-05-01:
OS: Ubuntu 16.04.6
GNU Radio: 3.8.1
Python: 3.8.2
SWIG: 4.0.1 (also tested 3.0.8)

A simple flowgraph with null sources and 2 Python sinks crashes with a segmentation fault.
The following flowgraph reproduces the issue. It contained 4 Python blocks (2 `my_null_source` blocks and 2 `my_null_sink` blocks) because I wanted to get rid of `gnuradio-blocks` in order to speed the build. I built GNU Radio using `cmake ../ -DCMAKE_BUILD_TYPE=Debug  -DPYTHON_EXECUTABLE=/usr/bin/python2.7 -DENABLE_DEFAULT=OFF -DENABLE_GNURADIO_RUNTIME=ON -DENABLE_PYTHON=ON -DENABLE_POSTINSTALL=OFF -DENABLE_DOXYGEN=OFF -DENABLE_VOLK=ON` in order to have the minimal build possible for running the test.
It reproduces with Python 3.8 on GNU Radio 3.8.* so I don't think it's a Python version issue.
This has been confirmed with GNU Radio 3.8 by Christophe Seguinot, Marcus Muller, and Volker Schroer. Volker also confirmed the crash happens with pybind instead of SWIG. They used different Ubuntu versions.

The error is a segmentation fault at `include/gnuradio/py_feval.h:69` which is the function `py_eval_ll::calleval(int)` and the line `return eval(x);`. Adding some debug prints shows the crash is not immediate (some function calls succeed before the crash).

I bisected the code and got to the following conclusions:

1. Old versions: `3.7.8git`, `3.7.9git`, `v3.7.8`, `v3.7.8.1`, and `v3.7.8.2` are all ok, but `v3.7.9rc1`, `v3.7.9`, `v3.7.9.3`, `v3.7.10`, `v3.7.10.1`, `v3.7.10.2`, and `v3.7.11` are not. Git commit `1206251231696359270a260508551e044f3af33a` seems to be the breaking change.
1. Newer versions: `v3.7.12.0`, `v3.7.13.0`, `v3.7.13.5`, and `v3.7.14.0` are all ok. Git commit `713629cce8d571570bc5f0f0db67c5a96d5ee071` seems to have fixed the bug in the 3.7 history (no idea why).
1. All GNU Radio 3.8 versions, starting at `3.8tech-preview`, and including current `master` (which is already 3.9), fail. 
1. GNU Radio 3.8 is a successor of `v3.7.12.0`, but its history contains the  commit `267d669eb21c514c18a6ee979f5cf247d251f1ad` ("Merge branch 'merge_next' (which merges next)" which says "There will be breakage"  (`v3.7.13.0` does not contain this commit).
1. Removing the references to `random.h` found in the breaking commit `1206251231696359270a260508551e044f3af33a` from `v3.8.1.0` source code does not solve the issue.

```
# -*- coding: utf-8 -*-
import os
import signal
import sys

from gnuradio import gr

import numpy

class my_null_source(gr.sync_block):
    def __init__(self, item=numpy.uint32, vlen=1):
        gr.sync_block.__init__(
            self,
            name="my_null_source",
            in_sig=None,
            out_sig=[(item, vlen)])

    def work(self, input_items, output_items):
        out = output_items[0]
        for i in range(len(out)):
            out[i] = 0

        return len(out)

class my_null_sink(gr.sync_block):
    def __init__(self, item=numpy.uint32, vlen=1):
        gr.sync_block.__init__(
            self,
            name="my_null_sink",
            in_sig=[(item, vlen)],
            out_sig=None)

    def work(self, input_items, output_items):
        return len(input_items[0])

class python_sink(gr.top_block):
    def __init__(self):
        gr.top_block.__init__(self, "Testing Python Sinks")
        # source1 = blocks.null_source(gr.sizeof_gr_complex)
        # source2 = blocks.null_source(gr.sizeof_gr_complex)
        source1 = my_null_source(item=numpy.complex64)
        source2 = my_null_source(item=numpy.complex64)
        # null_sink1 = blocks.null_sink(gr.sizeof_gr_complex)
        my_null_sink1 = my_null_sink(item=numpy.complex64)
        my_null_sink2 = my_null_sink(item=numpy.complex64)

        # self.connect(source1, null_sink1)
        self.connect(source1, my_null_sink1)
        self.connect(source2, my_null_sink2)

        print('GR Version: {}'.format(gr.version()))

        # print('Blocked waiting for GDB attach (pid = {})'.format(os.getpid()))
        # raw_input('Press Enter to continue: ')

def main(top_block_cls=python_sink, options=None):
    tb = top_block_cls()

    def sig_handler(sig=None, frame=None):
        tb.stop()
        tb.wait()

        sys.exit(0)

    signal.signal(signal.SIGINT, sig_handler)
    signal.signal(signal.SIGTERM, sig_handler)

    tb.start()

    tb.wait()

if __name__ == '__main__':
    main()

```

Full stack trace:

```
hread 20 "my_null_sink25" received signal SIGSEGV, Segmentation fault
#0  0x0000000002324b90 in ?? ()
#1  0x00007f5ec43df433 in gr::block_gateway_impl::work (this=this@entry=0x2324fc0, noutput_items=8,
    input_items=std::vector of length 1, capacity 1 = {...}, output_items=std::vector of length 0, capacity 0)
    at /home/user/gr/src/gnuradio/gnuradio-runtime/lib/block_gateway_impl.cc:139
#2  0x00007f5ec43df471 in gr::block_gateway_impl::general_work (this=0x2324fc0, noutput_items=<optimized out>, ninput_items=...,
    input_items=std::vector of length 1, capacity 1 = {...}, output_items=...)
    at /home/user/gr/src/gnuradio/gnuradio-runtime/lib/block_gateway_impl.cc:121
#3  0x00007f5ec43dc4c4 in gr::block_executor::run_one_iteration (this=this@entry=0x7f5e71ffadf0)
    at /home/user/gr/src/gnuradio/gnuradio-runtime/lib/block_executor.cc:514
#4  0x00007f5ec4439097 in gr::tpb_thread_body::tpb_thread_body (this=0x7f5e71ffadf0, block=..., start_sync=...,
    max_noutput_items=<optimized out>) at /home/user/gr/src/gnuradio/gnuradio-runtime/lib/tpb_thread_body.cc:122
#5  0x00007f5ec4428932 in gr::tpb_container::operator() (this=0x2ab6640)
    at /home/user/gr/src/gnuradio/gnuradio-runtime/lib/scheduler_tpb.cc:50
#6  gr::thread::thread_body_wrapper<gr::tpb_container>::operator() (this=0x2ab6640)
    at /home/user/gr/src/gnuradio/gnuradio-runtime/lib/../include/gnuradio/thread/thread_body_wrapper.h:52
#7  boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke (function_obj_ptr=...)
    at /usr/include/boost/function/function_template.hpp:159
#8  0x00007f5ec4444260 in boost::function0<void>::operator() (this=<optimized out>) at /usr/include/boost/function/function_template.hpp:773
#9  boost::detail::thread_data<boost::function0<void> const>::run (this=<optimized out>) at /usr/include/boost/thread/detail/thread.hpp:116
#10 0x00007f5ec27fc5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#11 0x00007f5ee913a6ba in start_thread (arg=0x7f5e71ffb700) at pthread_create.c:333
#12 0x00007f5ee831d41d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

(first reported at the mailing list, "Weird Python Sinks Crash with GNU Radio 3.8.1").

giladbeeri, 2020-05-01:
Christoph Mayer spotted that when the blocks aren't stored as local variables in `__init__()` (meaning, they are kept as class variables, so they don't get destroyed when `__init__()` ends), the crash doesn't reproduce.

I still don't have an idea why the commits I mentioned break and fix the issue, and why it only reproduces when there are at least 2 Python blocks.

hcab14, 2020-05-01:
> I still don't have an idea why the commits I mentioned break and fix the issue, and why it only reproduces when there are at least 2 Python blocks.

The problem is there with only one python block as well, as can be seen when running with address sanitizer.

giladbeeri, 2020-05-03:
To summarize, @hcab14 found that the underlying issue is with creating Python blocks that are only kept inside a function scope, then when the function ends, the Python objects are destroyed.

@marcusmueller do you consider it a bug in the framework?

giladbeeri, 2020-05-03:
> > I still don't have an idea why the commits I mentioned break and fix the issue, and why it only reproduces when there are at least 2 Python blocks.
> 
> The problem is there with only one python block as well, as can be seen when running with address sanitizer.

Yes, the memory corruption is there with only one Python block, but at least for me personally, the crash itself only reproduces when there are at least 2 Python blocks (I don't know why and it might be environment-specific).

marcusmueller, 2024-01-28:
This is too big an issue to tackle right now, but my guess is that the answer here is twofold:

1. Python blocks are still a bit problematic, because all Python blocks run in the same Python runtime, competing for locks and interaction with the GNU Radio runtime
2. The error at hand probably is some use-after-free with SWIG objects, which got solved through Pybind11 in 3.9; but that warrants deeper inspection.