gnuradio/gnuradio #3436: No module named _runtime_swig with Ubuntu 18.04 and 3.7.11 binaries

kgustke, 2020-05-01:
Are the install directions from https://wiki.gnuradio.org/index.php/InstallingGR still accurate?  Where is my dumb mistake?  I used to have this working on a previous Ubuntu 18.04 installation (last year).
Using fresh Ubuntu 18.04...
After installing Anaconda 3...  and setting up a python 2.7 environment (py27)

Installed from binaries:
$ sudo apt install gnuradio

$ gnuradio-config-info --version
3.7.11

$ gnuradio-config-info --prefix
/usr

$ gnuradio-config-info --enabled-components
python-support;testing-support;doxygen;sphinx;gnuradio-runtime;gr-ctrlport;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-atsc;gr-audio;* alsa;* oss;* jack;* portaudio;gr-comedi;gr-channels;gr-noaa;gr-pager;gr-qtgui;gr-trellis;gr-uhd;gr-utils;gr-video-sdl;gr-vocoder

$ source activate py27
$ export PYTHONPATH=/usr/lib/python2.7/dist-packages
$ echo $LD_LIBRARY_PATH
/usr/local/lib

$ python
Python 2.7.18 |Anaconda, Inc.| (default, Apr 23 2020, 22:42:48) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gnuradio
>>> from gnuradio import *
>>> from gnuradio import blocks
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/__init__.py", line 32, in <module>
    from blocks_swig import *
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig.py", line 22, in <module>
    from blocks_swig0 import *
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig0.py", line 17, in <module>
    _blocks_swig0 = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig0.py", line 16, in swig_import_helper
    return importlib.import_module('_blocks_swig0')
  File "/home/homeuser/anaconda3/envs/py27/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _blocks_swig0
>>> from gnuradio import gr
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/__init__.py", line 41, in <module>
    from runtime_swig import *
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 17, in <module>
    _runtime_swig = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_runtime_swig')
  File "/home/homeuser/anaconda3/envs/py27/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _runtime_swig

StephenCarlson, 2020-05-27:
Was this solved? I'm running into the same issue.

zirdisyukur, 2020-09-15:
getting this issue as well

kgustke, 2020-09-15:
My issue was "solved" for my use-case.  I got gnuradio working fully (and with UHD) by NOT installing Anaconda.  

Below is the install script that worked for me (from my notes)...

```
#!/bin/bash

# UHD wiki https://kb.ettus.com/UHD
#
# Use all three scripts to setup UHD and GNURadio for USRP Streaming with Ubuntu 18:
# 0_noanaconda_install_basics_1804_homeuser.sh
# 1_noanaconda_Install_Uhd_Ubuntu1804.sh
#  uhd_usrp_probe
##   linux; GNU C++ version 7.5.0; Boost_106501; UHD_003.010.003.HEAD-0-gef157678
##   FW Version: 5.1
##   FPGA Version: 33.0
# 2_noanaconda_install_gnuradio_1804_homeuser.sh
#  gnuradio-config-info --version
##  3.7.13.5
#https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux#Install_Linux#

#No Anaconda installed or environment used. That's what was messing me up'
sudo apt-get update

# This installs python 2.7 stuff and all the packages that show up in dist-packages
sudo apt-get -y install git swig cmake doxygen build-essential libboost-all-dev libtool libusb-1.0-0 libusb-1.0-0-dev libudev-dev libncurses5-dev libfftw3-bin libfftw3-dev libfftw3-doc libcppunit-1.14-0 libcppunit-dev libcppunit-doc ncurses-bin cpufrequtils python-numpy python-numpy-doc python-numpy-dbg python-scipy python-docutils qt4-bin-dbg qt4-default qt4-doc libqt4-dev libqt4-dev-bin python-qt4 python-qt4-dbg python-qt4-dev python-qt4-doc python-qt4-doc libqwt6abi1 libfftw3-bin libfftw3-dev libfftw3-doc ncurses-bin libncurses5 libncurses5-dev libncurses5-dbg libfontconfig1-dev libxrender-dev libpulse-dev swig g++ automake autoconf libtool python-dev libfftw3-dev libcppunit-dev libboost-all-dev libusb-dev libusb-1.0-0-dev fort77 libsdl1.2-dev python-wxgtk3.0 git libqt4-dev python-numpy ccache python-opengl libgsl-dev python-cheetah python-mako python-lxml doxygen qt4-default qt4-dev-tools libusb-1.0-0-dev libqwtplot3d-qt5-dev pyqt4-dev-tools python-qwt5-qt4 cmake git wget libxi-dev gtk2-engines-pixbuf r-base-dev python-tk liborc-0.4-0 liborc-0.4-dev libasound2-dev python-gtk2 libzmq3-dev libzmq5 python-requests python-sphinx libcomedi-dev python-zmq libqwt-dev libqwt6abi1 python-six libgps-dev libgps23 gpsd gpsd-clients python-gps python-setuptools
#AND PACKAGES THAT HAVE DRIVEN ME CRAZY WITH ANACONDA NOW WORK WITH PYTHON 

# NOW GET AND BUILD UHD
cd $HOME
mkdir workarea-uhd
cd workarea-uhd
git clone https://github.com/EttusResearch/uhd
cd uhd
git checkout release_003_010_003_000  # FPGA image 33!

git submodule update --init --recursive
cd host
mkdir build
cd build
cmake ../
grep -r -i  COMPAT_MAJOR ~/workarea-uhd/uhd | grep -i "0x\|#define"

# ./host/lib/usrp/x300/x300_fw_common.h:#define X300_FPGA_COMPAT_MAJOR 0x21
# make sure FPGA is 33 (0x21)
grep -r -i  X300_FPGA_COMPAT_MAJOR ~/workarea-uhd/uhd | grep -i 0x
make
make test
# 100% should pass
sudo make install
sudo ldconfig

##  Add LD_LIBRARY_PATH to .bashrc ##
sudo echo 'export LD_LIBRARY_PATH=/usr/local/lib' >> ~/.bashrc
##  Add PYTHONPATH to .bashrc ## 
sudo echo 'export PYTHONPATH=/usr/lib/python2.7/dist-packages' >> ~/.bashrc
sudo echo 'export UHD_RFNOC_DIR=/usr/local/share/uhd/rfnoc' >> ~/.bashrc
source ~/.bashrc

export PYTHONPATH=/usr/lib/python2.7/dist-packages

########### GnuRadio ##############
# If you want to use GNU Radio with a USRP, you FIRST must clone and install UHD. 
# https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux
#
# After UHD: Binary Install for gnuradio
# https://wiki.gnuradio.org/index.php/InstallingGR
#
sudo add-apt-repository ppa:gnuradio/gnuradio-releases-3.7
sudo apt update
cd ~
sudo apt -y install gnuradio
#sudo apt install gr-fosphor gr-osmosdr qtbase5-dev libqwt-qt5-dev   # Commented... no extra stuff

#Modtool on Ubuntu
#NOTE: On released builds for Ubuntu 18 (bionic), there is an issue using gr_modtool after GNU Radio has been installed from the PPA. This is due to byte-compiled code that remains in the modtool templates after installation. To workaround this issue: 
cd /usr/share/gnuradio/modtool/gr-newmod
sudo py3clean .
cd ~

# GNU radio should be installed and these tests should work:
gnuradio-config-info --version
gnuradio-config-info --prefix
gnuradio-config-info --enabled-components

# UHD files are in /usr/local/bin.  Example files are in /usr/share/gnuradio/examples/uhd/
# Gnuradio example files are in /usr/share/gnuradio/examples/
ln -s /usr/share/gnuradio/examples/ gnuradio-examples

ryanvolz, 2021-05-18:
I'm always excited when I stumble across a bug report for some random reason and then notice that I have something to add...

I think you've found the right solution here: don't mix environments from multiple package managers (e.g. `apt` and `conda`). I'm not an expert on everything Python does when looking for modules to import (is anyone?), but I highly suspect that the Anaconda python was not happy trying to load modules from the system's `/usr/lib/python2.7/dist-packages` and that more hackery with PYTHONPATH and LD_LIBRARY_PATH and perhaps others would be needed to make it work. So if you get `gnuradio` from `apt`, use `python` from `apt`. You could still have Anaconda installed, but don't choose the option to make it your default system Python.

Now I don't know why you were using Anaconda python to begin with, but I'll add a second option for anyone who still wants to use Anaconda with GNU Radio: install the `gnuradio` package through `conda`! You can do:
```
conda install -c conda-forge gnuradio
```
although I *highly* recommend installing into a purpose-built environment and not a global Anaconda `base` environment. See [the wiki](https://wiki.gnuradio.org/index.php/CondaInstall) for details or use [my installer](https://github.com/ryanvolz/radioconda). Note that only GNU Radio 3.8+ with Python 3 is available with `conda`.

closed 2021-05-18 by ryanvolz

3esmit, 2022-11-09:
Python is a great programming language, but it's env stuff is horrible. 

I am also having this issue, and in my opinion this shouldn't be so complicated. When using python software I always have this type of issues...