gnuradio/gnuradio #3437: USRP Source Unusable: len_tag_name error with USRP source

cclamb, 2020-05-01:
Everybody, just did a pybombs update this morning, and now I can't run anything. I get this error on len_tag_name when I try to generate a flowgraph:

>>> Generate Error: /home/user/grc/test_usrp_src_fail.grc: (NameError("'len_tag_name' is not defined"), 'uhd.usrp_source(\n...

I have eliminated the remainder of the trace to keep this short. Can supply if needed. Attached example flowgraph containing a stock USRP source connected to a null sink.

Was able to fix by removing the line:

             ),                                                                                                                                                                                                                                                                           
             % if len_tag_name:                                                                                                                                                                                                                                                           
             ${len_tag_name},                                                                                                                                                                                                                                                             
             % endif  

from uhd_usrp_source.block.yml

and replacing it with ')', but I'm not sure what the consequences of this change are. Thanks! I've attached a simple example showing the issue.

[test_usrp_src_fail.zip](https://github.com/gnuradio/gnuradio/files/4565838/test_usrp_src_fail.zip)

cclamb, 2020-05-01:
...also able to fix by adding this as a parameter to uhd_usrp_source.bloc.yml:

 -   id: len_tag_name                                                                                                                                                                                                                                                                     
     label: TSB tag name                                                                                                                                                                                                                                                                  
     dtype: string                                                                                                                                                                                                                                                                        
     hide: ${ 'none' if len(str(len_tag_name)) else 'part'}    

Currently at (git log --oneline):

6ad350b06 (HEAD -> maint-3.8, origin/maint-3.8) grc: use correct labels in log window context menu
17b5308a4 grc: Enable strict_undefined to ease Mako template debugging
3b86c6aa3 zeromq: Add bind argument for message based zeromq blocks.
a263ad80b filter: Fix firdes RRC filter gain for alpha == 1
...

cclamb, 2020-05-01:
So the above will fix the GRC issue where I can't gen a flowgraph, but then the resulting graph won't run; tried to tweak some of the template settings, but had no luck. Just went with the first possible fix.

drmpeg, 2020-05-08:
I've verified this bug. It's caused by commit https://github.com/gnuradio/gnuradio/commit/17b5308a406f5d2e6694d0913e82f30abd2844ca on maint-3.8 and commit https://github.com/gnuradio/gnuradio/commit/202cecbde26463bb5e2a0da1549a965a16d3b4d3 on master.

@haakov can you please investigate? The complete error is:
```
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "memory:0x7f7e15ce1630", line 56, in render_body
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 101, in __getitem__
    return compat_builtins.__dict__[key]
KeyError: 'len_tag_name'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/gnuradio-3.7.12git/lib/python3/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 64, in render
    return template.render(**namespace)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 462, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 838, in _render
    **_kwargs_for_callable(callable_, data))
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 873, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 899, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7f7e15ce1630", line 58, in render_body
NameError: 'len_tag_name' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/gnuradio-3.7.12git/lib/python3/dist-packages/gnuradio/grc/gui/PropsDialog.py", line 199, in update_gui
    self._update_docs_page()
  File "/opt/gnuradio-3.7.12git/lib/python3/dist-packages/gnuradio/grc/gui/PropsDialog.py", line 230, in _update_docs_page
    block_constructor = self._block.templates.render('make').rsplit('.', 2)[-1]
  File "/opt/gnuradio-3.7.12git/lib/python3/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 66, in render
    raise TemplateError(error, text)
gnuradio.grc.core.errors.TemplateError: (NameError("'len_tag_name' is not defined",), 'uhd.usrp_source(\n    ",".join((${dev_addr}, ${dev_args})),\n    uhd.stream_args(\n        cpu_format="${type}",\n        % if otw:\n        otw_format="${otw}",\n        % endif\n        % if stream_args:\n        args=${stream_args},\n        % endif\n        % if eval(stream_chans):\n        channels=${stream_chans},\n        % else:\n        channels=list(range(0,${nchan})),\n        % endif\n    ),\n    % if len_tag_name:\n    ${len_tag_name},\n    % endif\n)\n% if clock_rate():\nself.${id}.set_clock_rate(${clock_rate}, uhd.ALL_MBOARDS)\n% endif\n\n% if context.get(\'num_mboards\')() > 0:\n% if context.get(\'clock_source0\')():\nself.${id}.set_clock_source(${clock_source0}, 0)\n% endif\n% if context.get(\'time_source0\')():\nself.${id}.set_time_source(${time_source0}, 0)\n% endif\n% if context.get(\'sd_spec0\')():\nself.${id}.set_subdev_spec(${sd_spec0}, 0)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 1:\n% if context.get(\'clock_source1\')():\nself.${id}.set_clock_source(${clock_source1}, 1)\n% endif\n% if context.get(\'time_source1\')():\nself.${id}.set_time_source(${time_source1}, 1)\n% endif\n% if context.get(\'sd_spec1\')():\nself.${id}.set_subdev_spec(${sd_spec1}, 1)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 2:\n% if context.get(\'clock_source2\')():\nself.${id}.set_clock_source(${clock_source2}, 2)\n% endif\n% if context.get(\'time_source2\')():\nself.${id}.set_time_source(${time_source2}, 2)\n% endif\n% if context.get(\'sd_spec2\')():\nself.${id}.set_subdev_spec(${sd_spec2}, 2)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 3:\n% if context.get(\'clock_source3\')():\nself.${id}.set_clock_source(${clock_source3}, 3)\n% endif\n% if context.get(\'time_source3\')():\nself.${id}.set_time_source(${time_source3}, 3)\n% endif\n% if context.get(\'sd_spec3\')():\nself.${id}.set_subdev_spec(${sd_spec3}, 3)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 4:\n% if context.get(\'clock_source4\')():\nself.${id}.set_clock_source(${clock_source4}, 4)\n% endif\n% if context.get(\'time_source4\')():\nself.${id}.set_time_source(${time_source4}, 4)\n% endif\n% if context.get(\'sd_spec4\')():\nself.${id}.set_subdev_spec(${sd_spec4}, 4)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 5:\n% if context.get(\'clock_source5\')():\nself.${id}.set_clock_source(${clock_source5}, 5)\n% endif\n% if context.get(\'time_source5\')():\nself.${id}.set_time_source(${time_source5}, 5)\n% endif\n% if context.get(\'sd_spec5\')():\nself.${id}.set_subdev_spec(${sd_spec5}, 5)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 6:\n% if context.get(\'clock_source6\')():\nself.${id}.set_clock_source(${clock_source6}, 6)\n% endif\n% if context.get(\'time_source6\')():\nself.${id}.set_time_source(${time_source6}, 6)\n% endif\n% if context.get(\'sd_spec6\')():\nself.${id}.set_subdev_spec(${sd_spec6}, 6)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 7:\n% if context.get(\'clock_source7\')():\nself.${id}.set_clock_source(${clock_source7}, 7)\n% endif\n% if context.get(\'time_source7\')():\nself.${id}.set_time_source(${t [... cut ...] nself.${id}.set_gain(${gain31}, 31)\n% endif\n% endif  # if rx_agc31 != \'Enabled\'\n% if context.get(\'dc_offs_enb31\')():\nself.${id}.set_auto_dc_offset(${dc_offs_enb31}, 31)\n% endif\n% if context.get(\'iq_imbal_enb31\')():\nself.${id}.set_auto_iq_balance(${iq_imbal_enb31}, 31)\n% endif\n\n% if context.get(\'show_lo_controls\')():\nself.${id}.set_lo_source(${lo_source31}, uhd.ALL_LOS, 31)\nself.${id}.set_lo_export_enabled(${lo_export31}, uhd.ALL_LOS, 31)\n% endif\n% endif  # nchan > n\n')
```

haakov, 2020-05-08:
Yep, I'll take care of it.

haakov, 2020-05-08:
I don't have a gr-uhd-enabled GNU Radio build on my laptop at the moment, so I can't test it right away. But if anyone wants to try it out in the meantime, here's my initial guess for a fix: https://github.com/haakov/gnuradio/commit/413bfd9ae7e6d7b51589201bec0f646d9ed374a1

dkozel, 2020-05-08:
Did a very quick check, that worked for me on maint-3.8.

drmpeg, 2020-05-08:
I'm afraid the fix is not working for me. I did a clean build of Håkon's branch.

haakov, 2020-05-08:
@drmpeg Right, me neither. On second thought, there's no reason for the source to keep checking that variable if it's only defined for the sink. New PR here: https://github.com/gnuradio/gnuradio/pull/3464

cclamb, 2020-05-09:
Seems to be fixed, just updated my pybombs installation. Thanks everybody!

jdemel, 2020-05-12:
```
Failure
Traceback (most recent call last):
  File "memory:0x7fa60e06e880", line 72, in render_body
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 106, in __getitem__
    return compat_builtins.__dict__[key]
KeyError: 'i'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 64, in render
    return template.render(**namespace)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 476, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 878, in _render
    _render_context(
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 920, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 947, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7fa60e06e880", line 74, in render_body
NameError: 'i' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/gui/Application.py", line 710, in _handle_action
    generator.write()
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 83, in write
    for filename, data in self._build_python_code_from_template():
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 127, in _build_python_code_from_template
    blocks=self._blocks(),
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 197, in _blocks
    blocks = expr_utils.sort_objects(blocks, operator.attrgetter('name'), _get_block_sort_text)
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/utils/expr_utils.py", line 63, in sort_objects
    id2expr = {get_id(obj): get_expr(obj) for obj in objects}
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/utils/expr_utils.py", line 63, in <dictcomp>
    id2expr = {get_id(obj): get_expr(obj) for obj in objects}
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 184, in _get_block_sort_text
    code = block.templates.render('make').replace(block.name, ' ')
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 66, in render
    raise TemplateError(error, text)
gnuradio.grc.core.errors.TemplateError: (NameError("'i' is not defined"), '<%\n    all_options = [ option0, option1, option2, option3, option4 ][:int(num_opts)]\n    all_labels = [ label0, label1, label2, label3, label4 ][:int(num_opts)]\n%>\\\n# Create the options list\n% if int(num_opts):\nself._${id}_options = (\\\n% for ch in all_options:\n${ch}, \\\n% endfor\n)\n% else:\nself._${id}_options = ${options}\n% endif\n# Create the labels list\n% if int(num_opts):\nself._${id}_labels = (\\\n% for lbl in all_labels:\n% if lbl:\n${lbl}, \\\n% else:\nself._${id}_options[${i}], \\\n% endif\n% endfor\n)\n% elif labels:\nself._${id}_labels = ${labels}\n% else:\nself._${id}_map(str, self._${id}_options)\n% endif\n# Create the combo box\n% if widget == \'combo_box\':\n<%\n    win = \'self._%s_tool_bar\'%id\n%>\\\n${win} = Qt.QToolBar(self)\n${win}.addWidget(Qt.QLabel(${(label if (len(label) - 2 > 0) else repr(id))} + ": "))\nself._${id}_combo_box = Qt.QComboBox()\n${win}.addWidget(self._${id}_combo_box)\nfor _label in self._${id}_labels: self._${id}_combo_box.addItem(_label)\nself._${id}_callback = lambda i: Qt.QMetaObject.invokeMethod(self._${id}_combo_box, "setCurrentIndex", Qt.Q_ARG("int", self._${id}_options.index(i)))\nself._${id}_callback(self.${id})\nself._${id}_combo_box.currentIndexChanged.connect(\n    lambda i: self.set_${id}(self._${id}_options[i]))\n% endif\n# Create the radio buttons\n% if widget == \'radio_buttons\':\n<%\n    win = \'self._%s_group_box\'%id\n%>\\\n${win} = Qt.QGroupBox(${(label if (len(label) - 2 > 0) else repr(id))} + ": ")\nself._${id}_box = ${orient}()\nclass variable_chooser_button_group(Qt.QButtonGroup):\n    def __init__(self, parent=None):\n        Qt.QButtonGroup.__init__(self, parent)\n    @pyqtSlot(int)\n    def updateButtonChecked(self, button_id):\n        self.button(button_id).setChecked(True)\nself._${id}_button_group = variable_chooser_button_group()\n${win}.setLayout(self._${id}_box)\nfor i, _label in enumerate(self._${id}_labels):\n    radio_button = Qt.QRadioButton(_label)\n    self._${id}_box.addWidget(radio_button)\n    self._${id}_button_group.addButton(radio_button, i)\nself._${id}_callback = lambda i: Qt.QMetaObject.invokeMethod(self._${id}_button_group, "updateButtonChecked", Qt.Q_ARG("int", self._${id}_options.index(i)))\nself._${id}_callback(self.${id})\nself._${id}_button_group.buttonClicked[int].connect(\n    lambda i: self.set_${id}(self._${id}_options[i]))\n% endif\n${gui_hint() % win}')
```
Since @haakov pointed out this could be related, I thought, I'd share my error message. It only pops up with UHD blocks. In this very case with `gr-uhd/examples/uhd_fft.grc`.

closed 2020-05-21 by drmpeg

hellocaoziyi, 2020-05-26:
> ```
> Failure
> Traceback (most recent call last):
>   File "memory:0x7fa60e06e880", line 72, in render_body
>   File "/usr/lib/python3/dist-packages/mako/runtime.py", line 106, in __getitem__
>     return [...]

I met the same error, how can I solve this?

jdemel, 2020-05-26:
@hellocaoziyi This is an issue on GNURadio master branch. The fact that this issue is closed with a corresponding PR, that got merged, indicates that the latest GNU Radio master includes a fix for this issue. And in fact I just pulled the latest GR master and everything seems to be working again.
Thus, just pull the latest master and you should be fine. If you did not build GR master from source, you should write an email to the mailing list together with a lot of details how you ended up with your error.

hellocaoziyi, 2020-05-26:
> @hellocaoziyi This is an issue on GNURadio master branch. The fact that this issue is closed with a corresponding PR, that got merged, indicates that the latest GNU Radio master includes a fix for [...]

sorry to bother you, but my source is already latest, and the error shows NameError("'i' is not defined",), is the same as the error you show.

jdemel, 2020-05-26:
@hellocaoziyi no problem. 
A closed issue is probably not a good place to discuss your problem though. Please write an email to the mailing list. Or open a new issue where you explain your issue in detail.
Further, please provide more information on how you trigger your issue. e.g. your OS, how you installed GNU Radio, what exactly did you do to trigger your issue. I'd assume that'll help to narrow down your issue and it might trigger more specific questions to get to the bottom of your issue.

hellocaoziyi, 2020-05-26:
@jdemel could you give me your email address?

jdemel, 2020-05-26:
Please send your questions to <discuss-gnuradio@gnu.org>.
You can read more about the [GNU Radio Mailing List](https://wiki.gnuradio.org/index.php/MailingLists) in the GNU Radio Wiki.

hellocaoziyi, 2020-05-26:
my os: Ubuntu 18.04
the way I install GNU Radio: from source
version: maint-3.8
problem:
I use a example for https://github.com/bastibl/gr-ieee802-11.git named
"wifi_tx"
in fact, when I run examples include usrp, the error came out
---------------------------------------------------------------------------------------------------------------
Generating: '/home/webee/WEBee/WEBeeFile/wifi_tx_test.py'
Generate Error: (NameError("'i' is not defined",), '<%\n    all_options = [
option0, option1, option2, option3, option4 ][:int(num_opts)]\n
 all_labels = [ label0, label1, label2, label3, label4
][:int(num_opts)]\n%>\\\n# Create the options list\n% if
int(num_opts):\nself._${id}_options = (\\\n% for ch in all_options:\n${ch},
\\\n% endfor\n)\n% else:\nself._${id}_options = ${options}\n% endif\n#
Create the labels list\n% if int(num_opts):\nself._${id}_labels = (\\\n%
for lbl in all_labels:\n% if lbl:\n${lbl}, \\\n%
else:\nself._${id}_options[${i}], \\\n% endif\n% endfor\n)\n% elif
labels:\nself._${id}_labels = ${labels}\n% else:\nself._${id}_map(str,
self._${id}_options)\n% endif\n# Create the combo box\n% if widget ==
\'combo_box\':\n<%\n    win = \'self._%s_tool_bar\'%id\n%>\\\n${win} =
Qt.QToolBar(self)\n${win}.addWidget(Qt.QLabel(${(label if (len(label) - 2 >
0) else repr(id))} + ": "))\nself._${id}_combo_box =
Qt.QComboBox()\n${win}.addWidget(self._${id}_combo_box)\nfor _label in
self._${id}_labels:
self._${id}_combo_box.addItem(_label)\nself._${id}_callback = lambda i:
Qt.QMetaObject.invokeMethod(self._${id}_combo_box, "setCurrentIndex",
Qt.Q_ARG("int",
self._${id}_options.index(i)))\nself._${id}_callback(self.${id})\nself._${id}_combo_box.currentIndexChanged.connect(\n
   lambda i: self.set_${id}(self._${id}_options[i]))\n% endif\n# Create the
radio buttons\n% if widget == \'radio_buttons\':\n<%\n    win =
\'self._%s_group_box\'%id\n%>\\\n${win} = Qt.QGroupBox(${(label if
(len(label) - 2 > 0) else repr(id))} + ": ")\nself._${id}_box =
${orient}()\nclass variable_chooser_button_group(Qt.QButtonGroup):\n    def
__init__(self, parent=None):\n        Qt.QButtonGroup.__init__(self,
parent)\n    @pyqtSlot(int)\n    def updateButtonChecked(self,
button_id):\n
 self.button(button_id).setChecked(True)\nself._${id}_button_group =
variable_chooser_button_group()\n${win}.setLayout(self._${id}_box)\nfor i,
_label in enumerate(self._${id}_labels):\n    radio_button =
Qt.QRadioButton(_label)\n    self._${id}_box.addWidget(radio_button)\n
 self._${id}_button_group.addButton(radio_button, i)\nself._${id}_callback
= lambda i: Qt.QMetaObject.invokeMethod(self._${id}_button_group,
"updateButtonChecked", Qt.Q_ARG("int",
self._${id}_options.index(i)))\nself._${id}_callback(self.${id})\nself._${id}_button_group.buttonClicked[int].connect(\n
   lambda i: self.set_${id}(self._${id}_options[i]))\n% endif\n${gui_hint()
% win}')
>>> Failure
Traceback (most recent call last):
  File "memory:0x7f7dc2968c50", line 80, in render_body
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 101, in
__getitem__
    return compat_builtins.__dict__[key]
KeyError: 'i'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File
"/usr/local/lib/python3/dist-packages/gnuradio/grc/core/blocks/_templates.py",
line 75, in render
    return template.render(**namespace)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 462, in
render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 838, in
_render
    **_kwargs_for_callable(callable_, data))
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 873, in
_render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 899, in
_exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7f7dc2968c50", line 82, in render_body
NameError: 'i' is not defined

drmpeg, 2020-05-26:
I think this commit is missing from maint-3.8.

https://github.com/gnuradio/gnuradio/commit/154b2b73f22cc74940a536dbcc36f55ef636c3f7

hellocaoziyi, 2020-05-27:
@drmpeg you are right, my problem solved and I already submit this commit. thanks everyone.