gnuradio/gnuradio #3438: Log Power FFT

cclamb, 2020-05-01:
This generates an error:

Param - FFT Size(fft_size):
	Failed to eval ''part' if vlen == 1 else 'none'': name 'vlen' is not defined

This line in 

 -   id: fft_size                                                                                                     
     label: FFT Size                                                                                                  
     dtype: int                                                                                                       
     default: '1024'                                                                                                  
     hide: ${ 'part' if vlen == 1 else 'none' }    

is the cause, doesn't seem that the vlen variable exists.

haakov, 2020-05-01:
Fixed in master (but not in maint-3.8): https://github.com/gnuradio/gnuradio/pull/3130

dkozel, 2020-05-03:
@cclamb Thanks for posting the issue! @alekhgupta1441 has backported the fix and it's merged into maint-3.8 now.

PR #3444 (2020-05-03): gr-fft: fixup Log Power yml (Backport - 3.8)
Backport of PR #3130 to maint-3.8 to fix #3438
closed 2020-05-03 by dkozel